winder/Universal-G-Code-Sender #315: Problem with accents for the French language.

ArtFaks, 2015-11-26:
Hello, I'm french user and i've a problem with all accentuated characters after select the French language.

I use UGCS version 1.0.9, my OS is Windows 10 and my Java version is 1.8.0_66.

I've also a little problem with "Pendant". When I start the "Pendant" server, no IP is displayed in the console and I can't connect with my iPhone.

See screenshot
[image]

Jibeji, 2016-05-27:
Hi,
Same issue with version 2.0, there is an encoding problem.

winder, 2016-08-29:
I'm still seeing some strange localization issues with German, but French seems to work fine on an OSX machine.

winder, 2016-08-29:
@ml-hn I'm seeing some errors in POEditor as well. Things like `Schlieu00dfen` instead of `Schließen` can be fixed in the web interface.

There are a couple people who have been working on the German localization recently, if you have time to help you can join here: https://poeditor.com/join/project/2J2hB5I41Z

breiler, 2019-02-20:
Can't reproduce anymore, so I'm closing this.

closed 2019-02-20 by breiler