winder/Universal-G-Code-Sender #317: Why UGS of machine state in windows 7 ,Z position is covered by half?

viewsat, 2015-12-10:
[image]
Z position is covered by half

viewsat, 2016-02-18:
[image]

philreindl, 2016-02-20:
I haven't been able to reproduce this.  Do you know of anything that affects this?  I adjusted the windows font zoom, which didn't do anything.  What version of java are you running?  Have you changed the windows theme from the default?

jahnj0584, 2016-02-22:
I think your resolution is just too low. I know when I minimize it a bit I lose the Visualize/browse/save buttons.

viewsat, 2016-02-22:
I'm from Taiwan, we are using the traditional Chinese version of windows operating system, whether to display characters will be affected?

closed 2016-11-17 by winder