winder/Universal-G-Code-Sender #319: Router bit travels through the wood at fast speed before it begins carving

mza1979m, 2015-12-11:
I've been having this issue regularly and it's driving me nuts, and ruining my wood. I move the machine to where I want to begin, reset x, y & z to 0, confirm 100% that work position x, y & z are indeed all at 0. Then I begin by sending the code, and the first thing that happens (not all the time, but intermittently), is the router moves up and to the left (or right) about 2 inches in each direction, and simultaneously z moves down through the wood a few mm, creating a gouge in the wood before even beginning the carving. It does this travel at fast speed, as if it thinks it is above the wood. I use Fusion 360 as my CAM, and all the settings are good.

faultylee, 2015-12-16:
Could this be a Gcode problem? Might be useful check if the instruction is in fact from Gcode. I used to have similar problem as well with Gcode generated by some CAM software.

zarrstlouis, 2015-12-16:
I have had this same issue! zero'd machine, pressed send then craziness ensues... This only happens maybe 1 out 4 times, but it is a devastating issue. Also, this has only happened when Fusion 360 is generating my code.

This is the instruction lines from my gCode that was created from Fusion 360.

%
(Dust Shoe V2)
(T2  D=0.25 CR=0 - ZMIN=-0.665 - flat end mill)
G90 G94
G17
G20
G28 G91 Z0
G90

G28 is confusing me, as I am not setting a reference point other then zero. I am thinking either GRBL is not processing this command all the time or its a perimeter that I am somehow setting by mistake only sometimes....??? I also could be completely wrong and going down some mystery rabbit hole!! I am hoping someone with much more experience then me can help with this.

Thanks!

zarrstlouis, 2015-12-16:
I just resolved my issue in a post here and thought maybe it would help.  

https://discuss.inventables.com/t/universal-gcode-sender-fusion-360-issues/17368 

Have a great day!

mza1979m, 2015-12-18:
So I figured out the reason it was doing this. I don't really know anything about G28 or what you guys are talking about with regards to it. But what was happening is when I click "Send" in UGS, it would send the router to 0, 0, 0 of MACHINE POSITION (as opposed to WORK POSITION) first, and oftentimes Machine Position and Work Position are different. So the only way I've found was to do this:

1. Set the actual X and Y of the router physically to where I'd like the start position to be.
2. [Important] Set the Z position to ABOVE stock level by several mm.
3. Click "Close" in UGS to close the connection between the machine and the computer.
4. Click "Open" in UGS to reopen the connection. This resets Machine Position back to 0, 0, 0.
5. Lower the Z axis until it is sitting flush with the top of the stock.
6. Click "Reset Zero" to reset Work Position to 0, 0, 0.

Machine Position 'Z' should now be at a negative value (since we reconnected UGS to the machine while the the router bit was several mm above the stock.
Now when sending the Gcode, the machine goes to Machine Position 0 for 'Z'., which is several mm above stock, before going to the first cut.

Maybe this G28 solution is easier and makes more sense, who knows, but doing the steps above take seconds to do, since we're only doing 1-2 additional step to what we normally do.

zarrstlouis, 2015-12-18:
G28 is a return to home/reference command. In the post,  it describes,  if you jog to a home position, then send G28.1 in the the command area of your gcode sender, it will save. Or, you could just delete the G28 command from your gcode.

Hope this helps.

jahnj0584, 2016-09-28:
Will, this problem has been solved. It can be closed.

jahnj0584, 2016-09-28:
@winder

closed 2016-11-17 by winder