winder/Universal-G-Code-Sender #320: Warn when sending a file while machine isn't at 0,0,0

winder, 2015-12-13:
I'm working on multi-tool jobs for the first time and find that I keep accidentally sending a file before the machine has been zero'd out.

philreindl, 2016-01-15:
I've got limit switches set up on my machine, so I never start at 0,0,0.  GRBL is configured to home on startup, which moves to 0,0,0, then backs off slightly to get off the switches.  From there I'll jog to the work coordinates, set work coordinate 0,0, and then probe for z 0 (SUPER helpful with tool changes.)  Once the machine knows where 0,0,0 is, it doesn't matter much where you actually start, in my (limited) experience.  In summary, I'd prefer NOT to have this warning pop up.

winder, 2016-01-15:
Maybe UGS just needs better workflow support and this would be an annoying-for-a-lot-of-people bandaid. Yes, I think a plugin that loaded up multiple  gcode files with a little "tool change wizard" of sorts would go a long way towards helping with these sorts of mistakes.

closed 2016-01-15 by winder