winder/Universal-G-Code-Sender #322: GCode Stress Test

laserlars, 2015-12-14:
Hi there,

i use your tool for laser grayscale engraving which generates very, very large gcode files. I'd like to report a (memory) problem:

When sending a code with 1.000.000 lines, the UGS2nb works just fine.
However, when i try to send a full-hd image (50MB txt / over 2.000.000 lines), UGS will still send it row by row, but will stop every 500-2000 lines for a second (to flush/load lines, i guess), which causes the laser to completely burn the material there. Could you give me a hint what to try first?

i already added '-Xms1024m' to the start-windows.bat, which seemed to help a little... But when i set this higher, to -Xms2048m, UGS won't launch at all... :-(

thank you very much for your work, i hope you can help me out somehow!
if so, i'm totally gonna make a donation! :-)

winder, 2015-12-14:
I've been working on this for the past few weeks. See #302

You can check out the constant-memory branch to see progress on that. I haven't had a chance to do much testing, but the other night I successfully loaded a 1.6GB gcode file and got it to start sending. The gcode is going to be streamed from a file rather than loaded into memory.

The visualizer, console and response tables still need to be optimized though.

laserlars, 2015-12-14:
Ok, it was clearly my fault, sorry!!!
I somehow managed to run a 32bit java the whole time... 
With 64bit, UGS will get far more memory now :-)

A slight stuttering remains when i hit 'send',
but i guess you may close my issue :-)

closed 2015-12-14 by winder