winder/Universal-G-Code-Sender #327: Loopback endpoint

winder, 2016-01-15:
A "loopback" firmware endpoint for performance testing the UI would be useful so that users can make sure UGS wont choke on their gcode.

The main idea is that commands will go through the normal pipeline, but the serial connection layer is swapped out with a loopback connection which responds with "ok" and nothing else.

winder, 2016-01-15:
@philreindl I implemented a loopback endpoint tonight (in the constant-memory branch), basically just the way you suggested. It works great!

Interestingly, the change causes the master branch to crash, but the constant-memory work significantly simplified the UGS concurrency story so it works there with no problem.

philreindl, 2016-01-15:
Cool!  Funny thing, I flashed a spare arduino with grbl and have been doing UGS dev with that attached as you suggested, which also works great.  :)

commit 9f9ee502 (2016-01-15): Created a loopback endpoint for UI performance testing. Closes #327
commit 1e74d2af (2016-01-15): Created a loopback endpoint for UI performance testing. Closes #327
closed 2016-01-18 by winder