winder/Universal-G-Code-Sender #330: UGS 1.0.9 and 2.0 beta not working on MAC

Villafruela, 2016-01-19:
Hi guys,

I have mac os X 10.11.2 with shapeoko 2 and UGS 1.0.9 (tried 2.0.beta also) and last version of Java 8.

When i launch UGS & try to connect, i get this :
https://www.evernote.com/l/ABhjJtFVUiBFXpMdECxnws3p3kR88LlAo3IB/image.png

But as soon as i try to move i got the GRBL has not finished booting Error :
https://www.evernote.com/l/ABi5d4ITwopNuZRpnQoQvY3mfFalFnzi1uUB/image.png

I've seen that for others it was a baud rate problem but i'm at the correct 115200 and it does not work.

I have the same issue on both 1.0.9and 2.0 beta.

I can use some help here ;)

Thanks

winder, 2016-01-19:
It looks like the first part of the GRBL bootup message is getting cut off which is causing this problem.

There are two things you can try to workaround the problem:
1) Press the reset button on your arduino board and see if you get the full message.
2) Press the soft reset button in UGS, I believe GRBL re-sends the message.

I'm not sure why this would happen though, if anything I usually see extra stuff in the console from before GRBL starts up.

Do you have an older computer?
Or maybe a really long USB cable?

closed 2016-02-05 by winder

Villafruela, 2016-04-27:
Hi,

Sorry for the late answer, i forgot to keep you up to date ;)

Everything went up working after... i've rebooted my mac!

Thanks for your Help !