winder/Universal-G-Code-Sender #333: Visualization Issue

barnold96, 2016-01-21:
Since I work with inches, I changed UGS to report inches - $13=1 recently.  When running a job, inches are reported on the main window for X,Y,Z but the visualization is plotting in mm, apparently.  The tool icon barely moves; all the other visualization looks normal.

winder, 2016-01-22:
This is a known limitation to how the visualizer was implemented, basically the drawing code converts everything to MM as the gcode is parsed. Currently UGS doesn't have a good way of knowing what units GRBL is reporting, fortunately @philreindl is working on some improvements in this area that should help in a lot of areas - including the visualizer.

Here's one of the old tickets about this issue #96, which I thought was a duplicate of this but is actually describing a case that was fixed a long time ago (ironically, without the fix to that issue, the visualizer would have worked the way you expected).

barnold96, 2016-01-23:
Thank you for the update, Will.

--------------------------------------------

 

 
 
 it on GitHub.

philreindl, 2016-01-23:
The central theme of the work I was doing (which should be ready to merge, btw) is to isolate the jog gcode commands from the "real" gcode that's being run.  In order to do this we need to track units and motion mode.  As it sits, those things aren't exposed for the visualizer to use.  But I would expect it would be pretty straightforward to do so.

commit 27a3a9d8 (2016-02-18): Normalize visualizer tool position based on current units. Closes #333
closed 2016-02-18 by winder