winder/Universal-G-Code-Sender #335: UI Designer not matching what's actually built

philreindl, 2016-01-24:
I'm trying to implement a couple of fairly simple ui changes.  
1. Get rid of the commands tab, and move the command entry to the bottom of the console tab.
2. Add a current parser modal state indicator to the machine status area so it's clear what units the machine is parsing commands in.

My trouble is that the UI in the designer does not match what is shown when the program is actually running.  Obviously, this makes laying things out extremely difficult.
UI designer:
[image]

Actual:
[image]

Any suggestions as to what I'm doing wrong?  I'm using netbeans 8.1, installed just for this.  I haven't messed with any of the settings.  I'm using IntelliJ for everything but the form design.

winder, 2016-01-24:
The builder can be like this sometimes, it isn't always as WYSIWYG as it
should be.

Based on this: https://forums.netbeans.org/topic37380.html
And this: https://netbeans.org/kb/docs/java/gbcustomizer-basic.html

It looks like most of the issues can be resolved by choosing better layouts
(in the navigator window, right-click a JFrame and change in the `Set
Layout` section. Maybe a box layout would be simpler?

Also it looks like there is a new GridBagLayout builder that could help.

philreindl, 2016-01-24:
Hrm.

closed 2016-01-24 by philreindl