winder/Universal-G-Code-Sender #336: Auto connect

philreindl, 2016-01-24:
I've got a bit of code I've been playing with to auto connect.  Right now it's pretty aggressive at detecting disconnects and then reconnecting.  I think that's probably ok as long as the sender isn't streaming a file.  When a reconnect happens, the machine xyz is reset to zero, so at the very least we'd need to pause the stream and rehome.  But it's likely that some commands would be lost.  So it probably is best to just throw up an error in that case, cancel the job, and wait for user input.

While I don't personally have any motion commands set to happen on startup, I believe that it is possible.  So having the machine auto connect might not be desirable from that standpoint either.  

Any thoughts on this?  

I like the idea of having the machine auto connect on startup, it's one less thing to click with my crappy touchpad in the garage.  But the feature probably requires a new setting, which would be disabled by default.  I'd also like to see auto reconnect happen if the machine was connected, but got disconnected.  Again, this is probably a new setting, disabled by default.  If a disconnect happens when streaming a file throw up an error dialog and cancel the file.

winder, 2016-01-24:
A connect-on-startup feature sounds good, did you implement it in GUIBackend.java? I'm not sure about an auto-reconnect though, I'm having a hard time imagining what you could automatically do to recover from something like that.

Settings are pretty easy to add right now, just add a getter/setter in Settings.java and an initializer in SettingsFactory.loadSettings. The UGPlatform is using these settings as well and I'm not 100% happy about having both GUIs share the same settings (especially as new features are added to the Platform which don't get brought to the Classic GUI).

philreindl, 2016-01-25:
For me, it's pretty typical to run a job, shut off the machine entirely with the laptop and UGS still running.  It's a bit of a pain to get everything back up and running.  I'm hoping to be able to get the serial port to recognize the drop and properly reset without restarting everything.  So the autoreconnect is not intended to be something that would recover from a job that got messed up somehow.

As it sits, there's an autoconnect method on GUIBackend which attempts to connect using whatever was in the settings.  Before it's done, I'll have it scan the ports incase the serial port changes.  But baud and firmware shouldn't change, so those will be fixed.  The autoconnect method is being fired from a timer which is started in the main window.  I'd probably ultimately make GUIBacked own the timer, and would check the settings every time before attempting to autoconnect.

closed 2016-02-12 by philreindl