winder/Universal-G-Code-Sender #337: How to build/deploy the platform from within an ide

philreindl, 2016-01-26:
I see that the build is failing due to recent changes that broke the platform version of UGS.  From what I can tell, that guy depends on something from netbeans.  I haven't been able to figure out how to get it to run though (aside from downloading the artifacts.)  Is there any development docs I should be looking at to get started?

winder, 2016-01-26:
I've been working on documentation recently, but haven't gotten this far
yet (http://winder.github.io/ugs_website/)

If you open up the .travis file you can see the commands you need to run.
Basically you need to cope the latest .jar to a specific location, then you
can open the UGSPlatform directory with Netbeans and everything should
build.

philreindl, 2016-01-26:
Ah ok.  I was assuming there was something automatic happening there.

closed 2016-01-26 by philreindl

winder, 2016-01-26:
@philreindl I put together some information about building:
http://winder.github.io/ugs_website/dev/getting_started/

philreindl, 2016-01-26:
Thanks for the update.  Are you able to run the UGS Plaform from within your ide and attach a debugger?

winder, 2016-01-26:
Yes, its a little weird opening the project because you have to open the UGS project first. But it should import to netbeans just fine.

philreindl, 2016-01-27:
ok, I finally got it to go.  I didn't realize that maven was a separate install, so that was the first thing.  The second thing was the package job wasn't quite working for me.  Third, I didn't see that the jar file gets renamed in the copy command.