winder/Universal-G-Code-Sender #340: Timer Not Stopping

barnold96, 2016-01-30:
Loaded Jan 30 build this morning and ran a job.

Popup didn't appear when job ended.  Timer was still counting.  Hit "Cancel" and got this.

[image]

strider460, 2016-02-22:
Don't know if you have an idea already as to what's causing this issue so just say I have the same issue with USG 2.0 standard GUI and NOOBS on a RPI2. 

I'm trying to get an "enter" key press to press the send button on the GUI so having it get stuck just before the program ends is problematic...   

The problem is intermittent though, not every job does it (though I cant tell if it is effecting the timer, it appears to be accurate when it exits the program). Mine also does the same thing with the time being crazy wrong and sometimes negative when it does break. Gonna try to do my "enter" key thing on 1.0.9 and see if it was an issue then that just never really came up .

winder, 2016-02-22:
@strider460 I think this may be related to #347 so it could be fixed starting in the Feb. 21 build. Can you verify that you're using the latest code?

If anyone has a full console log from if this issue with a recent build, it could help. Thanks.

strider460, 2016-02-22:
If 2/21 was the most recent than I definitely wasn't up to date. I'll try it again tonight and if it throws a fit again I will get the console log.

strider460, 2016-02-23:
[Console Log.txt](https://github.com/winder/Universal-G-Code-Sender/files/141907/Console.Log.txt)
Console log attached in text file from RPi console while running latest UGS build

I'm not sure if I'm doing most of this right so here is what I ended up doing so if I'm making a rookie mistake I don't give bad info:

I re-downloaded the ZIP of the project and built the old GUI in NetBeans so I knew I had the most recent UGS. Opened the fresh UGS2.0 up on the Pi with "java -jar /filepath.jar/" to get console log (I'm hoping this is what you were looking for, this may be the rookie mistake) and ran a short file till it got stuck sending. Only took 2 or 3 tries;timer will keep counting up indefinitely, and numbers only go crazy negative like in the picture when you close the job.

strider460, 2016-02-23:
just opened the log and computer made a mess of the formatting. reformatted for clarity: 
[Console log2.txt](https://github.com/winder/Universal-G-Code-Sender/files/141920/Console.log2.txt)

winder, 2016-02-27:
@strider460 Unfortunately I don't see anything in the log. Could you send me a gcode file that you're seeing this problem with?

strider460, 2016-02-27:
[2-27logFile.zip](https://github.com/winder/Universal-G-Code-Sender/files/149542/2-27logFile.zip)
File has 2 test programs. one is the full program that has some startup and end G and M codes and the other file is just one G1 command followed by X Y moves from the same program. It also has the console log and the unaltered UGS2.0 build that I was using which looks like it is from 2-27.

winder, 2016-02-27:
@strider460 What version of GRBL are you using? Were you able to reproduce this using both of these files?

strider460, 2016-02-27:
yes, the "stuck sending" issue persists on both files. Just updated to 0.9j to test that idea with the same issue after a few "send"s in a row. It receives all of the ok's from GRBL so the job finishes but the "****file transfer complete****" message never appears.

winder, 2016-02-28:
If you use the command table all the commands have an ok?

strider460, 2016-02-28:
yes, all of the lines are checked "ok" in the command table when it gets stuck. I didn't catch that in this video but I got the console window for a working and non working "send":

https://www.youtube.com/watch?v=Qi7ZPYeecaw

winder, 2016-02-28:
@strider460 I managed to reproduce the issue, thanks for all the help. Stay tuned for a solution.

winder, 2016-02-28:
I've got a possible fix applied and updated the nightly build. Let me know if that helps.

winder, 2016-03-04:
@strider460 @barnold96 Were you guys able to verify that the issue is resolved?

barnold96, 2016-03-04:
The timer seems OK on the Jan.28 build that I'm using.  I'm not using the latest build because of the visualizer not working.

winder, 2016-03-05:
If the error is still happening, please reopen the ticket.

closed 2016-03-05 by winder