winder/Universal-G-Code-Sender #341: Startup

Willson13, 2016-02-05:
I have installed the latest Java update, as well as downloaded and unzipped the .zip file for version 1.0.7 and I cannot seem to get the window to open. I see a small black box flash on the screen, which I believe to be a command prompt window flashing on the screen, and nothing else happens. What do I do to fix this issue?

winder, 2016-02-05:
What operating system are you using?

Willson13, 2016-02-05:
I'm running windows 10. Is that going to be a problem?

winder, 2016-02-05:
Not a problem at all. Your question was how to get the file explorer to associate the .jar with Java which is different for every operating system.

This is a little unusual, because the java installer usually does that for you on Windows. It looks like you can find instructions to update file associations here: http://windows.microsoft.com/en-us/windows/change-default-programs#1TC=windows-7

You could also double click the `start-windows.bat` file.

closed 2016-02-05 by winder