winder/Universal-G-Code-Sender #344: Removing "Job complete after..." after fileStreamComplete()

strider460, 2016-02-17:
Question first: 
How hard would it be for someone not fluent in java to remove the "Job Completed After...." pop up after you send a file and set the main panels attention back to the "send" button instead of the "visualize" button (so that all you have to do is press enter to run again)?

Reason being:
I am working on a project at work and am trying to sell a shapoko or Xcarve as our answer. We are using the equivalent of a paste extruder to lay out some silicone for mounting buttons on an end product so the machine runs gcode for the path and uses Spindle on/off Mcode to turn the extruder on/off with a relay. It works great with my laptop hooked up but I am now hoping to get it running off a Raspberry Pi, and remove the splash screen after jobs so that when i get this ready for production all the guys on the floor have to do is place parts on the bed, hit enter, switch and repeat (I assume the splash screen is there to prevent exactly that, but its my only quarel with UGS for this project). I tied making a SD card streamer with another arduino over the TxRx lines, but i couldn't get the flow control to match UGS which I would be using to test any program I would put on the SD card anyway, so i like the idea of it running off of RPi. 

I'm new, be gentle:
I'm super new to java but I've been poking around in netbean trying to see if I can answer my own question but it doesn't look like I'm clever enough. I can see the  FileStreamComplete() in the MainWindow but Im not sure how to skip over displaying showMessageDialog() and resetting the attention to the send button without breaking stuff. 

Any help would be appreciated, I know the "Issues" section probably isn't the place to ask this sort of thing but I didn't know where else to ask. Thanks a bunch.

winder, 2016-02-17:
I never noticed how the button selection jumps around like that. Really strange.

I would take a slightly different approach though, in the "initProgram()" function add a keyboard shortcut so that enter always does the send:
```
case KeyEvent.VK_ENTER:
    if (sendButton.isEnabled()) {
        sendButtonActionPerformed(null);
    }
```

As for disabling the popup, you did all the heavy lifting on that one. Just comment out the call to `showMessageDialog`.

If you're really ambitious (maybe after you sell the company on a bootstrapped solution), you might consider making a UGS Platform plugin. You could have the UI show just the buttons you need and have keyboard shortcuts for them. I wrote up a tutorial for a small plugin with a custom window: http://winder.github.io/ugs_website/dev/plugin/

closed 2016-02-17 by winder

strider460, 2016-02-17:
Wow, that was awesome! Your "bootstrap" works just like I described and it only took you 10 minutes to respond. It took me two tries since I didn't scroll down through initProgram() and see all the other "case"s that your neat little shortcut should go next to but it works like a charm now. I will definitely check out your tutorial for Plugins, that looks like exactly what I'm after and I'm actually pretty excited now to get dirty with Java.

Thanks for making UGS so awesome!

winder, 2016-02-17:
Feel free to create more tickets like this if you have any more questions, especially if you start digging into the platform/plugins. The platform is still pretty new so I'm very interested to hear about first impressions.

montakhab, 2019-07-25:
hey guys
how can i remove completly UGS. i have crashed that window and i cant set that same as first day. I delated that and downloaded another but this one starts like previous. 
[image]