winder/Universal-G-Code-Sender #347: UGS used in the Raspberry pi [Error while stream starting file ....]

viewsat, 2016-02-18:
I use the Raspberry pi 2B test UGGS 2.0 Nightly (Feb 17,2016).
When I began to send the wrong message sometimes occurs [Error while stream starting file ....]
I have to do soft reset + $ x to return to normal send gcode.

[image]

viewsat, 2016-02-18:
[image]

winder, 2016-02-18:
Are you running any commands before pressing send?
How often do you see the problem?

viewsat, 2016-02-18:
1. Sometimes doing other commands (keyboard movement), and sometimes do not.
2. After the completion of gcode send again send gcode sometimes there are error
3. I often this problem.

sorry! my English is not good.

winder, 2016-02-18:
Could you try again with the latest build? I added a little more information to the error message.

viewsat, 2016-02-18:
ok

viewsat, 2016-02-19:
I try again with the latest build. The result was often error message occurred.

I opened this gcode test of 97 million lines to be tested.
http://www.picengrave.com/downloads/darklylabs/Roy%20Rogers%20Metric.zip

winder, 2016-02-19:
I added more information to the error message, does it say what the active commands are?

viewsat, 2016-02-20:
I tested the latest build error message occurs, you see the the video.

https://youtu.be/Q5wuohVWYvg

winder, 2016-02-20:
Thanks @viewsat that's what I needed. I'll take another look soon.

closed 2016-02-21 by winder

winder, 2016-02-21:
@viewsat this should be fixed now!

commit 2070ddef (2016-02-21): Reset active command buffer when a GRBL reset is detected. Closes #347
commit 96f8e890 (2016-02-21): Reset active command buffer when a GRBL reset is detected. Closes #347