winder/Universal-G-Code-Sender #350: Visualise Problem

PetriNogueira, 2016-02-18:
Hi everyone! I cannot use VISUALISE in the 1.9 and 2.0 versions. The software has a tilt and doesn't work anymore. I have close and open again, lost all machine positions. I use a notebook sony vaio with win10, intel core i3 2nd generation and 6GB RAM and JAVA always updated. Someone can help me? thanks a lot!

winder, 2016-02-18:
Please try running from the terminal, are there any errors reported?

PetriNogueira, 2016-02-18:
HI WINDER how I can do this? sorry the question.

winder, 2016-02-18:
Double click "start-windows.bat" and a black box with lots of text should open up behind universal gcode sender.

PetriNogueira, 2016-02-18:
Ok man... I will and talk about the results. THANKS!

PetriNogueira, 2016-02-18:
Nothing happens. I think could be a bug because I use Win10. Terminal sad: "Qt: Untested Windows version 10.0 detected!". When I click VISUALIZE the software just stop and no work.

winder, 2016-02-18:
UGS doesn't use Qt, so I'm not sure what this is.

Are you using the start-windows.bat file from UGS?
Is it in the same folder as the .jar?
What is the full path to the UGS jar file?
If you try moving the .jar file to C:\ does it work?

PetriNogueira, 2016-02-18:
Nothing... It's like a ghost maan!!! kkkkkkkkkkkkkkkkkkkk
The 1.8 version work perfect. 
I quit... kkkkkkkkkk

winder, 2016-02-18:
I can't help you if you quit.

It works on my Win 10 machine.

PetriNogueira, 2016-02-18:
KKKKKKKKKKK I quit because I don't know what to do, so I am using UGS 1.8. Antivirus, firewall or somethings like that can causing this problem?

winder, 2016-02-18:
Are you using the start-windows.bat file from UGS?

Is start-windows.bat in the same folder as the UniversalGcodeSender.jar?

What is the full path to the UniversalGcodeSender.jar?

If you try moving UniversalGcodeSender.jar to C:\ does it work?

PetriNogueira, 2016-02-18:
Answer your questions:

Are you using the start-windows.bat file from UGS? **yes**

Is start-windows.bat in the same folder as the UniversalGcodeSender.jar? **yes too**

What is the full path to the UniversalGcodeSender.jar?  
_C:\Users\Petri\Desktop\Projeto CNC\PROGRAMAS\UniversalGcodeSender-v2.0_
files: _README, start.sh, start-windows (.bat) and UniversalGcodeSender (jar executable)_

If you try moving UniversalGcodeSender.jar to C:\ does it work? **no**

PetriNogueira, 2016-02-18:
why you closed this issue?

winder, 2016-02-18:
When you double click start-windows.bat you should see a window with this at the top:

```
C:\Users\mac-vm\Desktop\with spaces>java -jar -Xmx256m UniversalGcodeSender.jar
Feb 18, 2016 11:27:44 AM com.willwinder.universalgcodesender.utils.SettingsFactory loadSettings
INFO: Loading settings.
Feb 18, 2016 11:27:44 AM com.willwinder.universalgcodesender.utils.SettingsFactory loadSettings
INFO: Settings file location: C:\Users\mac-vm\UniversalGcodeSender.json
```

Somewhere at the bottom should be a long error. I need the error.

PetriNogueira, 2016-02-18:
This is the complete code until I click VISUALIZE.

C:\Users\Petri\Desktop\Projeto CNC\PROGRAMAS\UniversalGcodeSender-v2.0>java -jar -Xmx256m UniversalGcodeSender.jar
fev 18, 2016 5:50:55 PM com.willwinder.universalgcodesender.utils.SettingsFactory loadSettings
INFORMAÃıES: Loading settings.
fev 18, 2016 5:50:55 PM com.willwinder.universalgcodesender.utils.SettingsFactory loadSettings
INFORMAÃıES: Settings file location: C:\Users\Petri\UniversalGcodeSender.json
fev 18, 2016 5:50:56 PM com.willwinder.universalgcodesender.model.GUIBackend applySettings
INFORMAÃıES: Applying settings.
fev 18, 2016 5:50:56 PM com.willwinder.universalgcodesender.model.GUIBackend addControllerListener
INFORMAÃıES: Adding controller listener.
fev 18, 2016 5:50:56 PM com.willwinder.universalgcodesender.model.GUIBackend addControlStateListener
INFORMAÃıES: Adding control state listener.
Qt: Untested Windows version 10.0 detected!
Qt: Untested Windows version 10.0 detected!
Qt: Untested Windows version 10.0 detected!
Qt: Untested Windows version 10.0 detected!
Qt: Untested Windows version 10.0 detected!
Qt: Untested Windows version 10.0 detected!
Qt: Untested Windows version 10.0 detected!
Qt: Untested Windows version 10.0 detected!
fev 18, 2016 5:52:25 PM com.willwinder.universalgcodesender.model.GUIBackend connect
INFORMAÃıES: Connecting to GRBL on port COM3
fev 18, 2016 5:52:25 PM com.willwinder.universalgcodesender.model.GUIBackend sendControlStateEvent
INFORMAÃıES: Sending control state change.
fev 18, 2016 5:52:25 PM com.willwinder.universalgcodesender.model.GUIBackend getGcodeFile
INFORMAÃıES: Getting gcode file.
fev 18, 2016 5:52:25 PM com.willwinder.universalgcodesender.model.GUIBackend getControlState
INFORMAÃıES: Getting control state.
fev 18, 2016 5:52:25 PM com.willwinder.universalgcodesender.model.GUIBackend getGcodeFile
INFORMAÃıES: Getting gcode file.
Qt: Untested Windows version 10.0 detected!
Qt: Untested Windows version 10.0 detected!
Qt: Untested Windows version 10.0 detected!
Qt: Untested Windows version 10.0 detected!
Qt: Untested Windows version 10.0 detected!
Qt: Untested Windows version 10.0 detected!
fev 18, 2016 5:52:51 PM com.willwinder.universalgcodesender.model.GUIBackend setGcodeFile
INFORMAÃıES: Setting gcode file.
fev 18, 2016 5:52:51 PM com.willwinder.universalgcodesender.model.GUIBackend initializeProcessedLines
INFORMAÃıES: Start preprocessing
fev 18, 2016 5:52:51 PM com.willwinder.universalgcodesender.model.GUIBackend getGcodeFile
INFORMAÃıES: Getting gcode file.
fev 18, 2016 5:52:51 PM com.willwinder.universalgcodesender.model.GUIBackend initializeProcessedLines
INFORMAÃıES: Took 10ms to preprocess
fev 18, 2016 5:52:51 PM com.willwinder.universalgcodesender.model.GUIBackend isConnected
INFORMAÃıES: Is connected: true
fev 18, 2016 5:52:51 PM com.willwinder.universalgcodesender.model.GUIBackend sendControlStateEvent
INFORMAÃıES: Sending control state change.
fev 18, 2016 5:52:51 PM com.willwinder.universalgcodesender.model.GUIBackend getGcodeFile
INFORMAÃıES: Getting gcode file.
fev 18, 2016 5:52:51 PM com.willwinder.universalgcodesender.model.GUIBackend getControlState
INFORMAÃıES: Getting control state.

winder, 2016-02-18:
Do those "Qt: Untested Windows version 10.0 detected!" messages come up when you try to open the visualizer?

PetriNogueira, 2016-02-18:
no. when I click VISUALIZE the bottom stay like pressing with the cursor and nothing happen, you know?

PetriNogueira, 2016-02-18:
how I send a picture here?

winder, 2016-02-18:
You can drag pictures into the text box

PetriNogueira, 2016-02-18:
This is before click VISUALIZE:
[image]

after:
[image]

and UGS doesn't work

winder, 2016-02-18:
Could you attach your .nc file? There might be something about it crashing the visualizer

PetriNogueira, 2016-02-18:
yeah, I can post here?
But, This happen with all .nc files...
I design in Inventor Pro 2016 with HSM

winder, 2016-02-18:
Yeah, try putting the file here. There might be something about Inventor Pro that breaks the visualizer.

winder, 2016-02-18:
You can also try one of the files here: https://github.com/winder/Universal-G-Code-Sender/tree/master/test_files

All of these should work, I use them for testing.

barnold96, 2016-02-23:
I'm just getting back in my shop after being on vacation.  I'm having the same issue after loading the latest nightly build.  I've tried with several gcode files I've used before, vut get no visualize window.

barnold96, 2016-02-24:
I ran a test on my laptop just now.  Using the Jan 24 build, Visualize works OK.  When I ran the latest build, I get the error posted above.  Here's the console output when running Jan 24:

[image]

barnold96, 2016-02-25:
I just downloaded and tried build 464.  Visualize still fails.  Apparently, the reason is a call to a .dll that doesn't exist.

**Exception in thread "AWT-EventQueue-0" java.lang.UnsatisfiedLinkError: Can't load library: C:\Users\BArnold\Documents\UniversalGcodeSender\natives\windows-amd64\\gluegen-rt.dll
**

I let the installation routine do its defaults that install to C:\Users\BArnold\Documents\UniversalGcodeSender.  No other folders are created.  As you can see above, an error is generated when UGS calls C:\Users\BArnold\Documents\UniversalGcodeSender\natives\windows-amd64\\gluegen-rt.dll.

..."\natives\windows-amd64\\gluegen-rt.dll" does not exist.

barnold96, 2016-02-29:
I downloaded and ran #467, Feb.28 nightly and still have the same issue as above.  Do I need to download gluegen-rt.dll and install it in addition to UGS?

barnold96, 2016-03-05:
Is anyone reviewing this thread?

barnold96, 2016-03-05:
I downloaded the latest nightly build this morning using the url link provided by Will.  The same error shows up as noted above.  Also, when first loading a gcode file, I get the following:

[image]

winder, 2016-03-05:
This should be resolved.

closed 2016-03-05 by winder

TinyMadHatter, 2016-06-16:
My black text box opens uo for a millisecond than closes automatically. 
I've installed Java, and now i'm installung Netbeens 7.1.0.2
Suggestions?
I've took the gost with the printscreen button but this is what i've captured:
[image]

winder, 2016-06-16:
@TinyMadHatter Looks like you're using a very very old version. Try downloading the latest: http://winder.github.io/ugs_website/download/

TinyMadHatter, 2016-06-16:
IT WORKED !!! Thank You!

hayttle, 2016-10-26:
I am using UGS 1.0.9 and same error.

PetriNogueira, 2016-10-26:
Hi!
I'm not solve this problem until now.
I change to bCNC. Best software! Amazing!

Now, Gcode Sender just to configure gbrl because it's more easy.

Enviado pelo meu dispositivo Windows Phone

Enviada em: ‎25/‎10/‎2016 23:37
Para: winder/Universal-G-Code-Sender<mailto:Universal-G-Code-Sender@noreply.github.com>
Cc: PetriNogueira<mailto:petrinogueira@outlook.com>; Author<mailto:author@noreply.github.com>

TheDoktar, 2018-12-13:
Hi Will, 

I've just downloaded and installed version 2.0, having previously tried an older version. 
I'm running on Ubuntu 16.04
I want to visualise without doing anything else right now so I set the connection port to /dev/null 
I run from the command line with the command
java -jar -Xmx256m ./UniversalGcodeSender.jar
and then load my file, which has a G91.1 as part of the initialisation, this appears not to be liked, so I commented that out. 
Now (and always in fact) when I click visualise I see a blank grey window, 
There is an exception on the command line as follows: 
<pre>
Dec 13, 2018 1:59:55 PM com.willwinder.universalgcodesender.model.GUIBackend getProcessedGcodeFile
INFO: Getting processed gcode file.
Exception in thread "AWT-EventQueue-0" java.lang.InternalError: XXX0 profile[1]: GL3bc -> profileImpl GL4bc !!! not mapped 
	at com.jogamp.opengl.GLProfile.computeProfileMap(GLProfile.java:2071)
	at com.jogamp.opengl.GLProfile.initProfilesForDeviceCritical(GLProfile.java:1954)
	at com.jogamp.opengl.GLProfile.initProfilesForDevice(GLProfile.java:1875)
	at com.jogamp.opengl.GLProfile.initProfilesForDefaultDevices(GLProfile.java:1842)
	at com.jogamp.opengl.GLProfile.access$000(GLProfile.java:80)
	at com.jogamp.opengl.GLProfile$1.run(GLProfile.java:230)
	at java.security.AccessController.doPrivileged(Native Method)
	at com.jogamp.opengl.GLProfile.initSingleton(GLProfile.java:216)
	at com.jogamp.opengl.GLProfile.getDefaultDevice(GLProfile.java:2027)
	at com.jogamp.opengl.awt.GLCanvas.<init>(GLCanvas.java:264)
	at com.jogamp.opengl.awt.GLCanvas.<init>(GLCanvas.java:231)
	at com.jogamp.opengl.awt.GLCanvas.<init>(GLCanvas.java:218)
	at com.willwinder.universalgcodesender.visualizer.VisualizerCanvas.<init>(VisualizerCanvas.java:146)
	at com.willwinder.universalgcodesender.visualizer.VisualizerWindow.<init>(VisualizerWindow.java:69)
	at com.willwinder.universalgcodesender.MainWindow.visualizeButtonActionPerformed(MainWindow.java:1284)
	at com.willwinder.universalgcodesender.MainWindow.access$2900(MainWindow.java:84)
	at com.willwinder.universalgcodesender.MainWindow$27.actionPerformed(MainWindow.java:980)
	at javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:2022)
	at javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2348)
	at javax.swing.DefaultButtonModel.fireActionPerformed(DefaultButtonModel.java:402)
	at javax.swing.DefaultButtonModel.setPressed(DefaultButtonModel.java:259)
	at javax.swing.plaf.basic.BasicButtonListener.mouseReleased(BasicButtonListener.java:252)
	at java.awt.Component.processMouseEvent(Component.java:6539)
	at javax.swing.JComponent.processMouseEvent(JComponent.java:3324)
	at java.awt.Component.processEvent(Component.java:6304)
	at java.awt.Container.processEvent(Container.java:2239)
	at java.awt.Component.dispatchEventImpl(Component.java:4889)
	at java.awt.Container.dispatchEventImpl(Container.java:2297)
	at java.awt.Component.dispatchEvent(Component.java:4711)
	at java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4904)
	at java.awt.LightweightDispatcher.processMouseEvent(Container.java:4535)
	at java.awt.LightweightDispatcher.dispatchEvent(Container.java:4476)
	at java.awt.Container.dispatchEventImpl(Container.java:2283)
	at java.awt.Window.dispatchEventImpl(Window.java:2746)
	at java.awt.Component.dispatchEvent(Component.java:4711)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:760)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:74)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:84)
	at java.awt.EventQueue$4.run(EventQueue.java:733)
	at java.awt.EventQueue$4.run(EventQueue.java:731)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:74)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:730)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:205)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)
</pre>
Unsure what this means. 

Any hints? 

regards

     Mark Winder. ( Yes, honestly. same surname!)

TheDoktar, 2018-12-13:
My gcode file starts like this: 
<PRE>
G17
#1 = 0.0500  ( Safe Z )
#2 = -0.0050  ( Engraving Depth Z )
G90
(( G91.1 ))
G20

F2.00
G0 Z#1
G0 X1.4161 Y0.1715
G1 Z-0.0000
G1 X1.4161 Y0.1715 Z-0.0000
G1 X1.4322 Y0.1843 Z-0.0200
G3 X1.3976 Y0.2467 I-0.4556 J-0.2124
G1 X1.3810 Y0.2456 Z-0.0066
G1 X1.3754 Y0.2414 Z-0.0031
G1 X1.3676 Y0.2355 Z-0.0041
</pre> 
.... etc