winder/Universal-G-Code-Sender #351: Homing at start-up?

hpladds, 2016-02-18:
If ( immediately after establishing a serial connection) I single click any of the manual controls for x y or z, my maker space's Shapeoko appears to go through a homing routine -- it makes as series of preset movements. Because we do not have limit switches, the cutting head occasionally runs out of bounds into the edges of the machine. 

This does not happened when running bCNC with the same config settings for Grbl.

It must be some sort of auto-homing routine, but what? 

Sapeoko 1
Grbl v 0.8
Gshield
Arduino Uno

hpladds, 2016-02-19:
No thoughts? Bump.

winder, 2016-02-19:
Do you have a startup block in grbl? https://github.com/grbl/grbl/wiki/Configuring-Grbl-v0.9#n---view-startup-blocks

What commands are listed in the UGS console?

hpladds, 2016-02-19:
I believe I discovered the problem and a work-around... of sorts

Synopsis of problem: After establishing a serial connection from Universal Gcode Sender (32 bit Linux) to Grbl 0.7 on an Arduino Uno / Shapeoko machine everything is fine until the "Manual Control" tabs are used to move to a zero position (relative to the work piece). 

I believe that the first tap of a "Manual Control" tab sends the following gcode (after a tap of the  Z+ tab for example): 

"G0 X0 Y0 Z1"

I wrongly assumed that this code implied that the X and Y axes would not move, but the Z would move up 1 step. I now think the machine endeavors to move to unknown/preset 0 (zero) locations for X and Y and also to a 1 (one) location for Z.

If, however, I click "Reset Coordinates" prior to clicking the "Manual Control" buttons, everything operates predictably. I suspect that hitting "Reset Coordinates" establishes that the current positioning is 0, 0, 0. When the "Manual Control" sends "G0 X0 Y0 Z1" to the machine, it only moves up one step. And does not seek some unknown/preset zero position.

winder, 2016-02-19:
Yikes, sounds like you're using a very old version of UGS. Download at least 1.0.9 (or the  Nightly Build). The current version uses relative movements rather than absolute movements.

Great analysis though, I forgot all about that issue from back when I first implemented the "Manual Control" feature.

closed 2016-02-19 by winder

hpladds, 2016-02-20:
Nope, but when I first read that section, I thought for sure that was it.
The machine was donated by my makers space, so the config files were a big
X factor.

hpladds, 2016-02-20:
donated to...

hpladds, 2016-02-20:
> Yikes, sounds like you're using a very old version of UGS. Download at
> least 1.0.9 (or the Nightly Build). The current version uses relative
> movements rather than absolute movements.
1.04 -- why I don't know. I'll update.

> Great analysis though, I forgot all about that issue from back when I
> first implemented the "Manual Control" feature.
>
> Mind me asking "relative to what?" Just a curiosity, this only happens
upon initialization, so where does the initial coordinate come from?

Thanks Again

winder, 2016-02-20:
Relative to the current location. See G91: http://linuxcnc.org/docs/html/gcode/g-code.html#gcode:g90-g91