winder/Universal-G-Code-Sender #355: UGS ignoring feed speeds?

jahnj0584, 2016-02-27:
So, I just realized it now. I have CAM to 4000mm/min, and said it'd take 2.5minutes to make this part. Ran it, seemed to be usual speed and took 6-7 minutes to really do, so no way it was doing the right thing. I also tried many y500 fXXXX commands in the console, and they are all the same speed! I also tried fXXXX y500 but that isn't any differenent. How do I ensure that it actually listens to my feed rates?!

V 1.09, can't use 2.0 because there is no console to send $X anymore (homing switches are in the mail) 

Cheers

winder, 2016-02-27:
UGS doesn't alter the feed rates unless you configured it to do so. You should check your GRBL settings, specifically [max rate ($110-$112)](https://github.com/grbl/grbl/wiki/Configuring-Grbl-v0.9#110-111-and-112--xyz-max-rate-mmmin) and [acceleration ($120-$122)](https://github.com/grbl/grbl/wiki/Configuring-Grbl-v0.9#120-121-122--xyz-acceleration-mmsec2). The defaults many preconfigured boards ship with are often set pretty conservatively.

closed 2016-02-27 by winder

jahnj0584, 2016-02-27:
>>> $$
$0=10 (step pulse, usec)
$1=25 (step idle delay, msec)
$2=0 (step port invert mask:00000000)
$3=6 (dir port invert mask:00000110)
$4=0 (step enable invert, bool)
$5=0 (limit pins invert, bool)
$6=0 (probe pin invert, bool)
$10=3 (status report mask:00000011)
$11=0.010 (junction deviation, mm)
$12=0.002 (arc tolerance, mm)
$13=0 (report inches, bool)
$20=0 (soft limits, bool)
$21=0 (hard limits, bool)
$22=0 (homing cycle, bool)
$23=0 (homing dir invert mask:00000000)
$24=25.000 (homing feed, mm/min)
$25=250.000 (homing seek, mm/min)
$26=250 (homing debounce, msec)
$27=1.000 (homing pull-off, mm)
$100=27.096 (x, step/mm)
$101=26.980 (y, step/mm)
$102=49.748 (z, step/mm)
$110=4500.000 (x max rate, mm/min)
$111=4500.000 (y max rate, mm/min)
$112=2500.000 (z max rate, mm/min)
$120=35.000 (x accel, mm/sec^2)
$121=70.000 (y accel, mm/sec^2)
$122=10.000 (z accel, mm/sec^2)
$130=346.000 (x max travel, mm)
$131=769.000 (y max travel, mm)
$132=45.000 (z max travel, mm)

Doesnt make sense. Is f3500 the correct command to move at 3500mm/min?

winder, 2016-02-27:
Yeah you should just need to add the feed command like you're doing. You can even verify by sending $G and seeing what the current GRBL feed rate is set to. The acceleration setting should control how fast it gets up to speed so you might need to increase that too.

jahnj0584, 2016-02-27:
I tried that, but if I do y700 the machine moves 700mm. When I do f500 y-700 to go back, it crashes against the end and obviously wants to travel more then it should?? 

>>> $G
[G0 G54 G17 G21 G90 G94 M0 M5 M9 T0 F500. S0.]

winder, 2016-02-27:
It's possible you're trying to drive the machine faster than the motors can handle. At some point the motors wont have enough power and will start losing steps.

jahnj0584, 2016-02-27:
but its going further then it should! And the speed of the movement is the same no matter what fX command I send with it, it just seems to want to go off into space if I add a fX command. The whole point of this is to SEE how fast my motors can go, but it keeps moving at the same no matter the FX command.

winder, 2016-02-27:
Oh I see you're problem here. You're in [G90 mode](http://linuxcnc.org/docs/2.6/html/gcode/gcode.html#sec:G90-G91), which is absolute coordinates.

If you start your machine at Y=0, you can go to Y=700. To go back you need to go to Y=0 not Y-500.

jahnj0584, 2016-02-27:
well, that solves ONE of the issues now. Feed rate still is unresponsive to any of the FX commands I send. There's no way its a lack of acceleration because it always gets up to the same speed, and stays there.

jahnj0584, 2016-02-27:
I mustve just switched it, because y-700 always did 700 back, not try to go 700 past my home.

jahnj0584, 2016-02-28:
Hey winder, i think it was something with my computer. Today, after a restart its reacting to feed speeds, and I was even able to edit a gcode from 3k to 3.8 and saw how it removed about 35% of the machining time, so it IS working now. Not sure what that was all about. Thanks for helping!

ChristianRenaux, 2016-11-18:
I do have exactly the same issue with the feed rate.  I am using Universal gcode sender 2.0.  
So whatever the command FXX the speed is always the same.  The only  way to change the speed is by modifying the $100-$101-$102 variables, which of course is not a good solution. 
Any idea ?

Skywalker200, 2018-06-25:
Hey guy's, I've been reading comments, I've been using Universal Gcode sender 2 for about 3 months, and I've notice that no matter what speed the programs say's to go, it's always running at full click, I realize this after braking a few bits. 

Winder says: 
UGS doesn't alter the feed rates unless you configured it to do so. You should check your GRBL settings, specifically max rate ($110-$112) and acceleration ($120-$122). The defaults many preconfigured boards ship with are often set pretty conservatively.
--

I don't get why all my *.nc file all have a feed rate that changes for cutting and for fast move, and what Winder is saying is that your scrude, and you have to change feed rate to the cutting rate for 110-112 for every different tool that you would be using.
What is the point of a programing a program to adjust the feed rate if the Universal Gcode sender can't even work with the adjustable feed rate needed for fast travels and for cutting. 

I've engraving a small dog tag, feed rate $110,$111,$120 had to be set to 40, it took over 8 hours to do one. That's a long time for one dog tag.

I really hope it has change, I other gcode sender that works with the feed and speed rate the program says to follow. I like to used this program, cause it has more features, but after what I read from Winder, I will look for another gcode sender, unless this problems I has been fix and I don't have the update.

I'm using a 2417 CNC  Laseraxe machine with Grbl 1.1f update.

Is this problem fix, and I;m missing and update?

carneeki, 2018-06-25:
Are you able to share a sample program that's not working the way you
expect? Perhaps a simple rectangle cut onto a dog tag?

Without more details, I'm going to start making some assumptions: It sounds
like whatever generates the G-code is using G0 to do linear moves rather
than G1 (G0 is used for rapid positioning at a maximum feed rate, such as
to move the cutter to near a starting point, while G1 will typically be
used for moving in lines, but at a feed rate specified by F<number>). What
software are you using to generate the G-code?

Skywalker200, 2018-07-04:
Hi, thanks for replying, I figure out the problem with the speed, the G54
what not in the code, even tho the definition of
G54 to G59 Work coordinate systems (WCSs) M T Have largely replaced
position register (G50
<https://en.wikipedia.org/wiki/G-code#G50_position_register> and G92
<https://en.wikipedia.org/wiki/G-code#G92_position_register>). Each tuple
of axis offsets relates program zero directly to machine zero. Standard is
6 tuples (G54 to G59), with optional extensibility to 48 more via G54.1 P1
to P48.

In no where in this definition does it state that it will fallow what ever
the program speed is and not full speed. If I don't have this G54 code in
my program, it's on full throttle all the way. You would think that it
would mention it in the definition.

But I still have another problem, first, I am engraving on dog tags,
second, it's starts fine, but close to have way through the job, it is no
long engraving but cutting through the entire piece. I run the exactly the
same gcode 5 time, and every time, I get a different result. I got it to
work fine on a demo piece, when I tried the same gcode to make the real
one, it's starts going to deep. Either 1/4,1/2 and 3/4 way through the job.
I've check the code, there is nothing about going deeper. I've check the
machine, there have been no jamming. I don't get it!

I'm sending you one of the gcode that I am trying to do, and the only
success was on a demo piece. When it comes to the real piece, some where,
something goes wrong.

> I forgot to ask... Can you also include your grbl settings by pasting the
> output of $$
>
> Thanks
>
>
> On Tue, 26 Jun 2018 at 08:22 Adam Carmichael <carneeki@carneeki.net>
>
> > Are you able to share a sample program that's not working the way you
> > expect? Perhaps a simple rectangle cut onto a dog tag?
> >
> > Without more details, I'm going to start making some assumptions: It
> > sounds like whatever generates the G-code is using G0 to do linear moves
> > rather than G1 (G0 is used for rapid positioning at a maximum feed rate,
> > such as to move the cutter to near a starting point, while G1 will
> > typically be used for moving in lines, but at a feed rate specified by
> > F<number>). What software are you using to generate the G-code?
> >
> >
> >> Hey guy's, I've been reading comments, I've been using Universal Gcode
> >> sender 2 for about 3 months, and I've notice that no matter what speed
> the
> >> programs say's to go, it's always running at full click, I realize this
> >> after braking a few bits.
> >> Winder says:
> >> UGS doesn't alter the feed rates unless you configured it to do so. You
> >> should check your GRBL settings, specifically max rate ($110-$112) and
> >> acceleration ($120-$122). The defaults many preconfigured boards ship
> with
> >> are often set pretty conservatively.
> >>
> >> I don't get why all my *.nc file all have a feed rate that changes for
> >> cutting and for fast move, and what Winder is saying is that your
> scrude,
> >> and you have to change feed rate to the cutting rate for 110-112 for
> every
> >> different tool that you would be using.
> >> What is the point of a programing a program to adjust the feed rate if
> >> the Universal Gcode sender can't even work with the adjustable feed rate
> >> needed for fast travels and for cutting.
> >>
> >> I've engraving a small dog tag, feed rate $110,$111,$120 had to be set
> to
> >> 40, it took over 8 hours to do one. That's a long time for one dog tag.
> >>
> >> I really hope it has change, I other gcode sender that works with the
> >> feed and speed rate the program says to follow. I like to used this
> >> program, cause it has more features, but after what I read from Winder,
> I
> >> will look for another gcode sender, unless this problems I has been fix
> and
> >> I don't have the update.
> >>
> >> I'm using a 2417 CNC Laseraxe machine with Grbl 1.1f update.
> >>
> >> Is this problem fix, and I;m missing and update?
> >>
> >> <https://github.com/winder/Universal-G-Code-Sender/
> issues/355#issuecomment-400042270>,
> DDgUffHlklGlks5uASXXgaJpZM4HkmvU>

**** Connected to COM6 @ 115200 baud ****
Grbl 1.1f ['$' for help]
>>> $$
$0 = 10    (Step pulse time, microseconds)
$1 = 25    (Step idle delay, milliseconds)
$2 = 0    (Step pulse invert, mask)
$3 = 4    (Step direction invert, mask)
$4 = 0    (Invert step enable pin, boolean)
$5 = 0    (Invert limit pins, boolean)
$6 = 0    (Invert probe pin, boolean)
$10 = 3    (Status report options, mask)
$11 = 0.020    (Junction deviation, millimeters)
$12 = 0.002    (Arc tolerance, millimeters)
$13 = 0    (Report in inches, boolean)
$20 = 0    (Soft limits enable, boolean)
$21 = 1    (Hard limits enable, boolean)
$22 = 0    (Homing cycle enable, boolean)
$23 = 0    (Homing direction invert, mask)
$24 = 50.000    (Homing locate feed rate, mm/min)
$25 = 635.000    (Homing search seek rate, mm/min)
$26 = 250    (Homing switch debounce delay, milliseconds)
$27 = 1.000    (Homing switch pull-off distance, millimeters)
$30 = 12000    (Maximum spindle speed, RPM)
$31 = 5000    (Minimum spindle speed, RPM)
$32 = 0    (Laser-mode enable, boolean)
$100 = 400.000    (X-axis travel resolution, step/mm)
$101 = 400.000    (Y-axis travel resolution, step/mm)
$102 = 400.000    (Z-axis travel resolution, step/mm)
$110 = 635.000    (X-axis maximum rate, mm/min)
$111 = 635.000    (Y-axis maximum rate, mm/min)
$112 = 635.000    (Z-axis maximum rate, mm/min)
$120 = 50.000    (X-axis acceleration, mm/sec^2)
$121 = 50.000    (Y-axis acceleration, mm/sec^2)
$122 = 50.000    (Z-axis acceleration, mm/sec^2)
$130 = 225.000    (X-axis maximum travel, millimeters)
$131 = 125.000    (Y-axis maximum travel, millimeters)
$132 = 170.000    (Z-axis maximum travel, millimeters)
ok
>>> $G
[GC:G0 G54 G17 G21 G90 G94 M5 M9 T0 F0 S0]
ok