winder/Universal-G-Code-Sender #356: (Suggestion) Auto connect to port COMX on start up

jahnj0584, 2016-03-02:
Easy as that. Let us choose in the settings which port to auto connect to when we launch UGS. Maybe even some other macro's afterwards, like insta-start the homing cycle OR launching a $X command for those without limit switches, but do a G0 L0 X0 Y0 Z0 to work 0 the machine on where it is.

winder, 2016-03-02:
Check the settings in the nightly build. :)

jahnj0584, 2016-03-02:
Mine only shows the option for Verbose console. Where exactly is this?

winder, 2016-03-02:
Oh you're using the platform, I still need to work on a settings GUI there.
If you load up the classic GUI you can enable a new auto connect feature.
Once enabled it should work in the platform as well.

jahnj0584, 2016-03-02:
Ah now I see it. Does speed override mean we can edit feed rates on the fly, or is this something else?
Maybe making a hottip box come after hovering over a setting for a few seconds to explain, like in HSMExpress? 

Cheers Winder. Thanks for the help/great program.

closed 2016-03-04 by winder

cpuwolf, 2023-05-20:
@winder I cannot find " a new auto connect feature" in classic GUI