winder/Universal-G-Code-Sender #357: (Suggestion) Stepper motor calibration plug in tool

jahnj0584, 2016-03-02:
Similar to Mach3. Assign a jog distance, set the $1 value to 255, and then jog. Asks for actual value, does the math of $10X * (Jog distance / real distance) and then sends the new $10X value to GRBL for us. So far in my experience it takes 2-3 runs to dial the machine in to the 0.01mm accuracy range. Repeat for the other 2 axis.

jahnj0584, 2016-04-16:
any ideas, will?

winder, 2016-04-25:
@jahnj0584 It's the kind of thing that would make a good plugin, but I don't have any plans to implement it right now.

jahnj0584, 2016-05-23:
if this is too much work, can we atleast do a calculator? I never use the macro buttons in Platform (mostly since theres no easy way to change them compared to the old versions). I could just pin it beside the jog control and do the math atleast in the same window as UGS.

breiler, 2018-05-20:
@jahnj0584 may I close this as it's now available in the setup wizard?

jahnj0584, 2018-05-20:
sure

closed 2018-05-20 by breiler