winder/Universal-G-Code-Sender #361: VISUALIZE is Still Broken

barnold96, 2016-03-06:
I just downloaded the latest nightly and ran a test.  It is still making a call to a non-existent file in a non-existent folder.

[image]
[image]

licensed2hench, 2016-03-07:
I am having the same issue with the 2.0 nightly build of the classic GUI.
I take it not everyone is? I am wondering is it something missing on my PC? A local Java issue?
Or perhaps some kind of permissions issue? 
I have tried it on a different PC and gotten the same error.
Here it the text copied from the console
[screen text.txt](https://github.com/winder/Universal-G-Code-Sender/files/161872/screen.text.txt)

licensed2hench, 2016-03-07:
Reading up on the "\ \" it looks like it has something to do with getting strings to display properly because single "\" are handled? But otherwise they act the same?
So as a test I copied the files inside the .jar from
UniversalGcodeSender\natives\windows-i586\natives\windows-i586
To
UniversalGcodeSender\natives\windows-i586\
And got the visualize button to work!
Looking at the other files in the natives folder it looks like the correct file structure.
Is this just a bad path statement? Or is visualize disabled for i586 users for now?

winder, 2016-03-07:
It isn't disabled intentionally. Theres something weird about JOGL and/or UGS build script causing an issue.

rugbymauri, 2016-03-07:
Same problem here.... 
Exception in thread "AWT-EventQueue-0" java.lang.UnsatisfiedLinkError: Can't load library: /home/mauri/apps/UniversalGcodeSender/natives/linux-amd64//libgluegen-rt.so   

seem that this library is missing

barnold96, 2016-03-08:
If UGS needs the gluegen-rt.dll library, shouldn't it load it as part of the installation process?

winder, 2016-03-08:
@barnold96 It is packaged correctly, as @licensed2hench commented. It just isn't working. If you want to take a look the build script is here: https://github.com/winder/Universal-G-Code-Sender/blob/master/pom.xml

The documentation is here: https://jogamp.org/wiki/index.php/JogAmp_JAR_File_Handling

When I have a chance I'll try downgrading back to 2.3.1, since I'm not sure what else could have changed at this point.

barnold96, 2016-03-08:
Is the creation of the folder "/natives/windows-amd64/" new in the latest version?  I've never seen that folder tree before.

licensed2hench, 2016-03-08:
Looking at an old build (1.0.9) the old path structure is:
UniversalGcodeSender.jar\natives\windows-i586\gluegen-rt.dll
not
UniversalGcodeSender.jar\natives\windows-i586\natives\windows-i586\gluegen-rt.dll
(and so on for all of the os and architectures)

So I am guessing the file structure is incorrect. Because all the different <os.and.arch> versions are seem to be affected I am guessing there is something up with the string variable?
${project.build.outputDirectory}
Wherever that "\ \" is coming from is possibly the culprit?

Comparing the build script to the documentation the first thing I notice is the build's scrip does not have the trailing "/" for any of them, but I don't know if that makes any difference.

Example:
Documentation:
<outputDirectory>/natives/windows-i586**/**</outputDirectory>

Build Script:
<outputDirectory>${project.build.outputDirectory}/natives/windows-i586</outputDirectory>

licensed2hench, 2016-03-08:
Could the output directory string already have the "natives / windows-i586 / " in it? 
That would explain where the duplicate "/ natives / windows-i586 "

And if the output string has the trailing "/" that would also explain the " / /"

winder, 2016-03-08:
@licensed2hench modifying '${project.build.outputDirectory}' didn't seem to
do anything to the natives directory in the jar file.

licensed2hench, 2016-03-08:
2 thoughts
1) try adding the trailing "/" as in the documentation. Maybe it needs to see that to properly terminate the function?
2) if that doesn't work put a "/test/' somewhere in the path of one just to try to lock down where the recursion / duplication is occurring?

jahnj0584, 2016-03-10:
I also have issues with visual in 2.0. Click the button, nothing comes up. Works fine in the Nov build.

shmorgan, 2016-03-14:
Same as jahnj0584

winder, 2016-03-14:
There is a new build up which hopefully resolves the problem. It's working on osx/win/Linux virtual machines.

I found there were some old build artifacts in my dev branch which is why this worked for me and no one else. So I was finally able to reproduce the issue after clearing out the build directory.

barnold96, 2016-03-15:
I just downloaded the latest build and ran a test.  Visualize seems fine now and also is tracing in inches, which is the way I have grbl set to report.

Good work!

barnold96, 2016-03-15:
One issue I see with the latest update is that I get a "GRBL hasn't finished booting" error window on first connect.  When I close the connection and re-open it, all is fine.

shmorgan, 2016-03-17:
Looks good, seems to be working now.

winder, 2016-03-18:
Thanks for the feedback and testing everyone!

closed 2016-03-18 by winder