winder/Universal-G-Code-Sender #364: (Offer) German Translation

jahnj0584, 2016-03-14:
My family is German, and I speak it nearly fluently. If you want to translate it, let me know a)if you want that and b)how I do it properly so its in the program.

cheers

winder, 2016-03-14:
I get to reference the documentation for the first time :)
http://winder.github.io/ugs_website/contrib/translation

The UGS Platform build isn't quite ready for translation. I'm trying to cleanup the last bits for the first official release (little things... like fixing the visualizer). Once things stabilize a bit I'll figure out the localization strategy.

winder, 2016-05-09:
I've finally got the platform to be localizable (most of it, anyhow). I'm trying out a new service to manage localization maintenance, if you're still willing the link to sign up is here: http://winder.github.io/ugs_website/contrib/translation/

closed 2016-05-09 by winder