winder/Universal-G-Code-Sender #370: [Help Requested] Visualizer Color Suggestions

winder, 2016-03-21:
I've been working on UI preferences with the UGS Platform and just got visualizer colors hooked up.

I'm not super happy with the defaults I came up. If anyone wants to contribute suggestions, there are preferences now so give it a spin and show us what you come up with.

[image]

philreindl, 2016-03-22:
Not exactly a color suggestion, but one of the things I've been thinking would be nice is if the spindle speed could be reflected in the visualizer.  I'm using a laser, controlled by setting spindle speed for intensity, which results in variations in the amount of burn when rastering.  It would be awesome if the visualizer could give an indication as to what the result will be.  The linear movements would be a gradient rather than a fixed color.

winder, 2016-03-22:
@philreindl A laser module would be awesome. The platform renderer is split up into a bunch of classes extending "Renderable" so that it can be extended by modules. For this specific request, there could be some "color line by speed" boolean that changes how `GcodeModel:updateVertexBuffers` colors the lines. The feed speed is already a property even, it just isn't used.

I'm trying to polish things up for an inaugural 2.0 release, then I'll start working on fun things like this (a box joint generator, dovetail generator, auto leveler, etc).

Eventually I'll get a killer feature into the platform release and win you over. ;)

jahnj0584, 2016-03-22:
Red X axis
Green Y Axis
Blue Z Axis
Dark Grey tool marker
Light color background, not plain white
med blue tool cutting path
Yellow jogging/rapid paths

Will you be adding camera controls? I'd enjoy arrow key panning, and maybe a top+bottom/left+right/front+back preset camera buttons.

winder, 2016-03-22:
You can zoom/pan with the mouse wheel and shift-drag. The keyboard controls are still working if you undock the window, I'm not sure whats swallowing up the commands before they get to the visualizer.

jahnj0584, 2016-03-22:
Here's mine now. Looks very similar to my CAM which is nice. A feature to invert the mouse rotation would be nice!

philreindl, 2016-03-22:
@winder _Eventually I'll get a killer feature into the platform release and win you over. ;)_
haha Well so far I've found the friction involved in developing on the platform to be higher than I'm willing to tolerate.  Part of the problem is that I'm a huge fan of IntelliJ, which just isn't practical for the netbeans UI platform.  The fact that the core module has to be compiled separately also makes things harder than they should be, IMO.  Again, my personal prejudice and not a dig on what you're doing.  For now, I'm content to keep picking at the monolithic legacy version and attempt to win you back.

winder, 2016-03-22:
@philreindl I agree with you about the friction, the core app used to be tied to netbeans as well. Somehow there's a way to mavenize the Netbeans platform build which would help. I've made a couple stabs at it, but have yet to get it working.

closed 2016-04-10 by winder

OslayViera, 2024-04-25:
> @philreindl A laser module would be awesome. The platform renderer is split up into a bunch of classes extending "Renderable" so that it can be extended by modules. For this specific request, there [...]

I already know that it is a closed topic but I am having the same need and I don't know if at some point the color by speed option was created and no matter how much I search I can't find how to do it

breiler, 2024-04-26:
Make sure that you are using a recent version of UGS (preferrably 2.1.6). There is not much documentation on this, but you can read about it in the PR that I made a while back: https://github.com/winder/Universal-G-Code-Sender/pull/2241

OslayViera, 2024-05-08:
Apparently it doesn't work with the F codes that my machine uses :(
[image]

breiler, 2024-05-08:
@OslayViera What is the use case for visualizing the feed rate?

OslayViera, 2024-05-08:
@breiler I have an X and Y axis CNC machine to make pyrography on wood with a nichrome tip depending on the feed speed. UGS is the only application that works perfectly with the GCode model I work with, the only need is the viewer, I would like to be able to see exactly the progress on the image and not on a colored rectangle. I don't know if this CNC model is the only one of its kind that uses this type of feed or this type of GCode

breiler, 2024-05-09:
@OslayViera I have added a new setting for the feed rate min/max colors which will be blended with the spindle speed min/max colors: https://github.com/winder/Universal-G-Code-Sender/pull/2515

It is available in the nightly builds here:
https://github.com/winder/Universal-G-Code-Sender/releases/tag/nightly

OslayViera, 2024-05-12:
[image]
@breiler It works perfect, I don't know how you did it so quickly but thanks for the urgency 👌🏻

breiler, 2024-05-12:
Awesome @OslayViera, if you get the chance I would love to see some pictures of both your machine and any pyrography you've made.