winder/Universal-G-Code-Sender #374: New GUI to big for 1024*768 screen

rugbymauri, 2016-03-24:
Hi the new GUI is to big

[image]

 my proposal is to put to Connection oanel on the top, like this

[image]

philreindl, 2016-03-25:
Crud, you're right.  

My plan to tighten things up is to have the connection panel only show when the machine is disconnected.  When connected, it will be replaced by the machine status panel (which will get the close button.)  Having them both onscreen at the same time seems redundant to me.

I think we can get rid of a row of the buttons in the file section as well by being a bit clever.  Send/pause can be combined.  I still don't see what good Save is.  Cancel and browse can probably be combined.  Visualize can be a tab in the area on the right.

Enable keyboard movement should be a setting.  For that to work though, jog should be disabled whenever the cursor is in a text box.  (file dialog/ console/ macro edit)

rugbymauri, 2016-03-25:
Visualize can be implemented as menu item -> view -> visualize

rugbymauri, 2016-03-25:
You can put all connection options (port, baud rate....) in a tab. And leave just the connect button. Normally the connection options dont change.

jahnj0584, 2016-03-28:
I realized this yesterday too. Turns out my monitor can go higher, but I find myself going back to the classic platform as the new one has issues, like none of the common command buttons being usable and a failed attempt at macro keys.

winder, 2016-03-28:
@jahnj0584 please create tickets for any issue you see in the platform - it's under heavy development right now.

rugbymauri, 2016-03-29:
I made a Mockup of possible changes to the Gui

- Connect Menu (no menu items), like a button (tricky to find the right event. MouseClick?)
- Visulize Menu (no Menu Items), like a button (tricky to find the right event. MouseClick?)
- moved Connection Pane in the Tabs. used only once to setup the connection,
- stripped some buttons as proposed by @philreindl 

now it should fit in small screens 

[image]

philreindl, 2016-03-29:
I'm working on making the status panels on the left a bit more dynamic.  The idea will be to hide the current machine status panel when the machine is NOT connected, and hide the bulk of the connection panel when the machine IS connected (Still need the disconnect button).  I'd like to give similar treatment to the file window.

My hope is to:
A. Make things fit properly on a small screen 1024x768 for sure, I'm planning to get a small tablet to run this on eventually, which may be more in the 800x600 range. 
B. Reduce the amount of UI navigation for "normal" use.  Settings adjustment, macro editing, etc are not what I'd group with "normal" use.  So those functions would be buried a bit from the main screen.
C.  Make the overall flow of what needs to happen next more intuitive, especially for new users.  First, need to connect, Second need to home (if enabled), Third load a file, Fourth run the file, Fifth, see how much time you  have to eat snacks while it runs.

philreindl, 2016-03-29:
@fautlylee mentioned using a PC at 1024x600

winder, 2016-03-29:
@philreindl What do you think about "forking" MainWindow.java (copying/pasting it to a new class) so that the old UI is stable while your in experiment mode? The nightly build gets 50-100 downloads a day.

At one point I was trying to support 640x480 since thats what the original raspberry pi supported out of the box. I'm not sure how widely that configuration is used though.

philreindl, 2016-03-29:
That's not a bad idea at all.  I had no idea nightlys had that much traffic.

philreindl, 2016-03-29:
@winder Actually, rather than just adding another MainWindow to the codebase, what do you think of getting the edge branch up to date and having it generate nightly builds as well?  Getting it up to date with master should be easy since there's no unmerged commits there.  I'm not sure what's involved with hooking it up to the build system.

faultylee, 2016-03-29:
Kudos to all of you for catering to edge use cases :+1:

PR #379 (2016-03-31): Put the jog panel back where it was.  
Screen works at 1024x600.  Resolves #374 

[image]
commit 39db73e3 (2016-03-31): Put the jog panel back where it was.  Screen works at 1024x600.  Resolves #374
closed 2016-03-31 by winder