winder/Universal-G-Code-Sender #378: Feature Request: Additional TinyG Support

licensed2hench, 2016-03-30:
Add the "Pause", "Resume" and "Cancel" button function in the "File section" for the TinyG firmware.

I am in the process of switching over to a TinyG controller and just discovered that most of the file and Machine Control buttons are not supported. Presumably because the commands these buttons would send are different in TinyG that in GRBL.
I have no idea how much work it would be to have the buttons change their functionality based on the firmware selected. If it is a major undertaking to code I would ask that at least the File area buttons of Pause / Resume and Cancel be added. The rest in the Machine Control are area not a big deal as they can be manually entered or set up macros.
Some reference info if needed:
https://github.com/synthetos/TinyG/wiki/TinyG-Feedhold-and-Resume

winder, 2016-03-30:
This isn't a big undertaking at all, there are unimplemented methods in [TinyGController.java](https://github.com/winder/Universal-G-Code-Sender/blob/master/src/com/willwinder/universalgcodesender/TinyGController.java) which need to be implemented with the TinyG specific handlers.

Compare the unimplemented methods to the [GrblController](https://github.com/winder/Universal-G-Code-Sender/blob/master/src/com/willwinder/universalgcodesender/GrblController.java).

licensed2hench, 2016-03-30:
Good to hear. :)
So a "TinyGUtils.class" would need to be created with the proper TinyG commands? 

Hmm, looking at the "GrblUtils" it looks like I was talking about the "real time" commands
It looks like PAUSE and RESUME are the same for both GRBL and TingG.
Status is {"sr":null}. so would that be {"sr":null} or {"sr":} or{"sr"}?
And reset... seems to be the same as GRBL, ctrl-X (0x18)

TINYG_PAUSE_COMMAND = '!'; 
TINYG_RESUME_COMMAND = '~'; 
TINYG_STATUS_COMMAND = '{"sr":null}'; 
TINYG__RESET_COMMAND = 0x18;

licensed2hench, 2016-03-30:
Looking at the Non real time commands:
It looks like TinyG has hard and soft alarms
https://github.com/synthetos/TinyG/wiki/TinyG-Alarms
Clearing a soft alarm {"clear":true} or $clear=1 
Clearing a hard alarm requires a RESET, control X (^x) or ( 0x18)

It looks like TinyG can give a lot of information back via the status report command '$sr' or '?' (for plain text response)
https://github.com/synthetos/TinyG/wiki/TinyG-Status-Reports
but it appears it returns all the statuses not just the one. So some kind of filter function may need to be applied? I don't know is done with the returned data. :/
Check G Code mode - ?
View parser state - ?

view TinyG settings, I think this is the same as GRBL, '$$' ?

TINYG_KILL_ALARM_LOCK_COMMAND = "$clear=1"; 
TINYG_TOGGLE_CHECK_MODE_COMMAND = "?";
TINYG_VIEW_PARSER_STATE_COMMAND = "?";
TINYG__VIEW_SETTINGS_COMMAND = "$$"; 

I wish I knew how to compile java stuff so I could mod and test this stuff out. I can make simple changes to existing jar files, but new code is beyond my skill level.

licensed2hench, 2016-03-30:
Looking at the GCode Commands
The only one that looks like it should be different is the homing cycle? 
G28.2 (G28.3 resets the machine zero without moving)
https://github.com/synthetos/TinyG/wiki/Homing-and-Limits-Description-and-Operation

TinyG_RESET_COORDINATES_TO_ZERO= "G10 P0 L20 X0 Y0 Z0"; 
TinyG_RESET_COORDINATE_TO_ZERO = "G10 P0 L20 %c0"; 
TinyG_RETURN_TO_ZERO_LOCATION= "G91 G0 X0 Y0 Z0"; 
TinyG_RETURN_TO_MAX_Z_LOCATION = "G90 G0 Z"; 
TinyG_PERFORM_HOMING_CYCLE= "G28.2 X0 Y0 Z0";

winder, 2016-03-30:
If you install NetBeans, just open up the project. There is a green "Play" triangle which will build the code and run it.

You don't really need the `TinyGUtils` class, just put something like this in place of `throw new UnsupportedOperationException("Not supported yet.");
` 
```
if (this.isCommOpen()) {
   sendByteImmediately((byte)'!');
      or
   sendCommandImmediately("G91 G0 X0 Y0 Z0");
}
```

There's some more documentation here: http://winder.github.io/ugs_website/dev/getting_started/

closed 2016-05-14 by winder