winder/Universal-G-Code-Sender #380: Jumbled Display

barnold96, 2016-04-02:
I downloaded the latest passing build a short while ago and this is what I see:

[image]

Backed up to build 500 for a normal looking display.

philreindl, 2016-04-02:
What resolution is that?  If you make the window wider does it look better?

winder, 2016-04-03:
I think the minimum sizes just need to be tweaked to make sure you can't shrink the window too much.

philreindl, 2016-04-04:
What is the "official" minimum supported window size?  We should probably set that as the minimum size on the window itself.

winder, 2016-04-04:
VGA was the lowest I ever actively tried to support (640x480), but 1024×768 is probably reasonable.

philreindl, 2016-04-04:
1024x768 is more or less what I figured. Going off topic a bit.. Have you thought much about incorporating any usage statistics reporting (Google Analytics)?  Would be really nice to have concrete data to back things like this up.

winder, 2016-04-04:
I've considered it. I was planning to look at what was build into the netbeans platform, but maybe something more general would be useful.

For an estimate, I have google analytics hooked up to the new website. Here are the statistics for the download page (not filtering out mobile devices):

[image]

philreindl, 2016-04-04:
Thanks for sharing.  That's an interesting mix.  If you're already using Google Analytics, it might make sense to stick with that.  https://developers.google.com/api-client-library/java/apis/analytics/v3

barnold96, 2016-04-04:
I'm back in my shop today and checked some of the different recent passing builds.  The layout that looks more balanced is 501.  I see what's happening with 514 and window size but the layout is clunky.  Sorry I didn't try resizing before.

faultylee, 2016-04-07:
@winder I think GA might give false result, at least in my case. I run UGCS on a dedicated machine, so I would download using my laptop and then copy the file over, instead of having a browser installed on the machine running UGCS :)

Like @philreindl said, having some form of analytic on the software might be helpful, though if the machine running it is not connected to the internet, then there might be another issue. And there might be noise coming from those testing the nightly on a different machine.

commit 7f6ced63 (2016-04-10): Remove the macro button panel from the MainWindow, this should hopefully come back in the near future. Closes #380
closed 2016-04-10 by winder