winder/Universal-G-Code-Sender #383: Feed Override

jahnj0584, 2016-04-06:
Either Im using it wrong, or it's currently not working. If I enable it and set it to 60 or 120 or 500 my machine is operating exactly the same. Does it only apply to new jobs? Is it as a multiplier, where 100 is regular speed, 50 is half and 200 is twice as fast? Does it effect the plunge, ramp, rapid, and cutting feeds? Something seems to be off here Wilder, and I'd love if it was a working function as I do the same jobs out of different woods and don't want to make 5 different CAM files for each species (ie Ebony needs to be done 30% slower then MDF)

Cheers

winder, 2016-04-10:
When you open a file the whole file gets processed right away, so you'll need to re-open the file after changing any settings.

It's possible it wasn't working on the platform build until recently, but I tried it just now and it's working. Let me know if you still see any problems.

closed 2016-04-10 by winder

jahnj0584, 2016-04-10:
ok cool. I'll keep that in mind next time. But what about the mutlipliers? I loaded quite a few files with it enabled and at 500 or 100 or 60 and they were still all acting as if it wasnt a factor and just behaved as I had it programmed.

jahnj0584, 2016-04-10:
I will install the latest version tomorrow on my machine and try it all again for you. Thanks.