winder/Universal-G-Code-Sender #392: (Suggestion) Seperate jog sizes for the XY and Z axis

jahnj0584, 2016-04-16:
Lets be honest, we're not jogging our Z by 50mm so how about 2 controls, one for each? That way I can arrow key in 50mm increments for XY, and pgdown/pgup 5mm for my Z. Maybe even a multiplier option? (Like as I just gave as an example, Z moves 1/10th as the other 2 axis)

There are also a few other issues right now Will, like the common macro window still being disabled even if you connect to your arduino in the latest build. Please fix!

Cheers,

philreindl, 2016-04-17:
For the macro window issue, can you please provide more details?  Are you using the classic or platform gui?

jahnj0584, 2016-04-25:
I forget actually. Ill download the latest version and see if its still there.

jahnj0584, 2016-04-27:
macro still doesnt work. Opened another thread for it.

jahnj0584, 2016-05-09:
still want to see this implemented

winder, 2016-05-14:
Just noticed this is a duplicate of #217

closed 2016-05-14 by winder