winder/Universal-G-Code-Sender #399: What is the "optimal" UGS Platform GUI screen resolution  ?

splitn2, 2016-04-26:
Hi, I have looked through the wiki and issues and I cannot find any info on the gui resolution. I have a RPi with UGS latest platform installed, and also a previous Platform version too. Keen to get it going for some testing with my wee GRBL. Is anyone able to give me some guidelines on what the best screen resolution is and what the GUI has been designed to work with? Thanks, Richard ps. keep up the great work team I love this project, thanks for your generosity with your open source efforts!

winder, 2016-04-26:
You should be able to use it with a low resolution but might have to move
the Windows around. Let me know how it works, this ui is still pretty new
so I'm always looking for feedback.

splitn2, 2016-04-26:
Thanks @winder I am using this wee screen which is quite neat with its touch screen but it is only 800x600. Yup I'll do some more playing with it and report back in due course...... cheers

closed 2016-04-29 by winder