winder/Universal-G-Code-Sender #400: Fonts very very small in high resolution monitor

simonmtl, 2016-04-28:
I'm using a windows 10 based laptop with QHD+ resolution (3,200 x 1,800). The fonts are so small they can barely be
[UGS 1.09 on QHD plus resolution.pdf](https://github.com/winder/Universal-G-Code-Sender/files/241409/UGS.1.09.on.QHD.plus.resolution.pdf)
 seen.  Microsoft says this is caused by the use of local fonts instead of the system ones.

simonmtl, 2016-04-28:
I just discovered USG platform, which solves my problem. Case closed  for me.

winder, 2016-04-29:
This is a known issue for Java Swing applications in general. Glad to hear the platform build works, I didn't realize it had high dpi handling. Java9 is supposed to have native handling for application scaling, but its still a ways out.

closed 2016-04-29 by winder