winder/Universal-G-Code-Sender #403: 'No serial ports found.'  on Mac OSX10.10.5 Yosemite

hobbietime, 2016-05-06:
Just downloaded the latest version 1.0.9
executed the .jar file... it loads perfectly except for the message
'No Serial Ports Found'

The ports are there... I use them all the time with Adruino IDE or CoolTerm...
however they do not appear with this software? ( no other comm software is running)

any ideas would be greatly appreciated.
Thanks in advance

winder, 2016-05-06:
Are you using an Arduino clone? There is a bug in the serial library that prevents it from listing some clone devices. If thats the case, you can enter the port address manually.

hobbietime, 2016-05-06:
using Arduino clone using wchusbserial1410 / 115200 8-N-1
How do I input the port address manually?   
I have tried to input    wchusbserial1410   directly and nothing happens

hobbietime, 2016-05-06:
Error opening connection (jssc.SerialPortException): Port name -wchusbserial1410; Method name -openPort(); Exception type - Port not found.

winder, 2016-05-06:
Try `/dev/tty.wchusbserial1410`

hobbietime, 2016-05-06:
perfet... that did the trick.... thank you very much...  just document in the realm file...
cheers

closed 2016-05-06 by winder

hobbietime, 2016-05-06:
sorry  the ReadMe   file

VelhoLobo, 2019-12-04:
Bom dia Srs., tenho o mesmo problema, macOS Mojave, Versão 10.14.16
**"Error opening connection (jss.SerialPortException): Port name - /dev/cu.whcusbserial410; Method name - openPort(); Exception type - Port no foud."**
Desde já agradeço penhoradamenete, qualquer ajuda
Helcio Xavier