winder/Universal-G-Code-Sender #405: Issues with T5 M6 commands?

jahnj0584, 2016-05-09:
Seems to be causing issues with the newer versions. Never was an issue before, just started now that I upgraded. Removing the line from each gcode is a bother with all the work and operations I do! Please fix, maybe just add an option to skip or auto delete these? 

Or, teach me to edit the post processor to stop adding these. Whatever is easiest. I use HSMworks.

winder, 2016-05-09:
This is a feature actually. In current version of GRBL lines with an error on them are completely ignored. Not good if you have a command like "G21 G90 G64 G40" at the beginning of a file to set modes.

Previously UGS would silently ignore these errors.

A quick google search suggests that a GRBL post processor is available for HWMworks. I don't have any experience with that software though.

closed 2016-05-09 by winder

weirdguy79, 2016-05-09:
I don't use solid works and have no idea about post processing

winder, 2016-05-09:
@weirdguy79 For any CAM software you need to use the appropriate post processor, GRBL is widely supported at this point.

jahnj0584, 2016-05-09:
yes, i know theres a processor, im wondering if theres a way to edit it to not include M and T commands in the code?

winder, 2016-05-09:
@jahnj0584 I'll see if it's possible to allow the job to continue when it halts like this, maybe adding some sort of "at your own risk" warning.

jahnj0584, 2016-05-09:
Please do. I'd appreciate it. I don't want to start new threads, but maybe add an option to "never show again" for the "report to github" splash when you start, and I occasionally get errors that say "error while processing response <ok>; null" when I do lots of jogging. Cheers