winder/Universal-G-Code-Sender #409: CNC stop moving during sending a file

tema2121, 2016-05-11:
Hi, 
I have new X-carve 1000x1000 
Using Mac (old one with 128 mb video memory) OSX El Capitan  
I try to mill a simple job. 
I can connect just at 115200 baud. 
I have tried all versions of Gcode sender and it always stuck at the beginning of milling. 
When I try to move I need often do a Soft Reset because cnc not responsible each time after 5 pressings on machine control moving buttons.  

and in a console alway appear some thing like that: 
idle.WPos 0000.00000..... MPos.... 0000.0000
Is it issue of my Mac or Arduino 
Please help me, it is a week since I try to find a solution.

winder, 2016-05-11:
Could you share the file you're trying to send?

tema2121, 2016-05-12:
[Krepez-300mm.zip](https://github.com/winder/Universal-G-Code-Sender/files/260824/Krepez-300mm.zip)

tema2121, 2016-05-12:
I have 4 gb of RAM,  so I have encreased memory in the start.sh file

tema2121, 2016-05-12:
[image]

tema2121, 2016-05-12:
Here is a code, after it cnc not response.

tema2121, 2016-05-12:
may be it can help you

winder, 2016-05-12:
@tema2121 This works fine for me with v1.0.9 and the nightly build. Please try upgrading UGS 1.0.7 is 2 years old.

closed 2016-05-12 by winder

tema2121, 2016-05-12:
I have tried all versions, how to encase memory for that application on OSX.
Is it ok if I have just 128 mb Video card on my old mac?

tema2121, 2016-05-13:
should I send grbl sketch to my Arduino Uno before using Universal Gcode sender.
Because when I try ti check firmware - it listed - firmware not found.