winder/Universal-G-Code-Sender #411: Platform considers jogs as lines from file

jahnj0584, 2016-05-13:
No idea why, but whenever I jog to do a 000 alignment, with each jog a line in the visualizer changes from red to white (loaded to sent). Doesnt effect the end result, but should be fixed! I had to reload the file for the visualizer to change everything back to red lines. 

thanks

winder, 2016-05-13:
Nice catch, thanks.

jahnj0584, 2016-05-13:
thanks. I am doing a production of 8 guitar necks, each one has 6+ operations to use. As you can imagine I spend a lot of time using this, and will probably find a lot of the bugs we have to iron out. Im really loving where the platform is going right now. 1 or 2 more features, and no bugs and it'd be perfect for me.

winder, 2016-05-13:
Check out the "workflow" plugin, I made it as a proof of concept but it could help manage the multiple files

jahnj0584, 2016-05-13:
I did actually, but it wont help me. My bed only fits 3 at a time, so I just run the first 2 (same file) then as the 3rd is running i load 2 more planks onto my bed and just keep going. Every operation requires a new fixture, bit, 000, or part of wood so it doesnt help me out too much. Thanks though!

jahnj0584, 2016-09-28:
issue is still there with latest version.

closed 2018-04-13 by winder