winder/Universal-G-Code-Sender #412: (Suggestion) Make "Work Position" the value of what commands are sent, not where it physically is.

jahnj0584, 2016-05-13:
Lets say you send 3 commands of x100 after hitting the reset home button. Your work position should say x300 right away, instead of having it load up as your machine moves. Machine position can stay real time. This would help when you are doing a hold down jog over 1 (where the machine vs screen are not moving at same speed) and could know when to stop sending commands.

Cheers

winder, 2016-05-14:
There's not much I can do about this, the value in these fields come directly from the controller.

closed 2016-05-14 by winder

jahnj0584, 2016-05-15:
alrighty, it was worth a try. Sorry for flooding you this week hahaha. 6 hours a day of machine time leave lots of time to think of improvements. Maybe i'll just get a capacitor to stop false hard stops from triggering and re enable them.

chamnit, 2016-05-15:
@jahnj0584: Coincidentally, this is one of the last remaining issues I intend to fix in Grbl v1.0, but it's turned out to be fairly tricky to do with Grbls planner in the way. One idea is to force a planner sync, where motions are emptied from the queue every time a work coordinate offset is changed. Not ideal, but it'd work.

chamnit, 2016-05-15:
Oops. Misread the issue. Your suggestions isn't hard to implement, but only useful for jogging. Hopefully this new jogging mode I'm installing will fix this issue in a different way.

jahnj0584, 2016-05-15:
yea, i basically want the jogging value to show what has been sent, not what has been moved. Make sense? Another example would be if my y is jogging 500mm, and i send a 200X command as well it'd show that on my work position as Y600 X200, but machine would still be Y(moving) X(0)