winder/Universal-G-Code-Sender #414: Options on homing axis order

jahnj0584, 2016-05-14:
Sometimes, dropping the Z then going over to 0,0 doesnt work regarding hold downs and screws and all that (as well as some programs being 0'd from the table). It'd be cool if we can choose if the Z goes to 0 before or after the X/Y
Cheers

sssfox, 2016-05-26:
I agree with this comment.
the current order makes the RETURN TO HOME button useless.
When I push it, it drags the tool across the surface of the material.
If you could just change the order so that Z moves last, it would be great.
Steve Fox

jahnj0584, 2016-05-26:
if thats happening steve, you should be setting your z0 just a hair over the material, to a point where a piece of paper gets caught between the bit and the material.

TheWoodButcherer, 2016-06-29:
I also agree. This is a much needed fix.

The bit dragging across the work piece can either ruin the piece or if using a fine bit, they often break.

My work around has been to jog back as close as possible to X0Y0 and then hit return to zero.

I cannot see that reversing the order of the code for the button click should be so hard.

commit 36a28d93 (2016-07-31): Stop using z-lift for return to zero. Closes #414, #308, #274, #263
closed 2016-07-31 by winder