winder/Universal-G-Code-Sender #415: Is it compatible with OSX EL Capitan? I see weird massages in a console.

tema2121, 2016-05-14:
Hi, 
that wired massages are interrupt work of Universal Gcode Sender always including during sending.
It looks like: MPos 00000,000 WPos. 0000.00000
I know exactly that it something between my laptop and Universal Gcode Sender 
I have OSX EL Capitan. on others computers it works.

Thanks a lot for your future help.
Artem
[image]

tema2121, 2016-05-14:
[image]
[image]
[image]

tema2121, 2016-05-14:
On others computers there are no such feedback massages. They interrupt communication. 
Thanks 
Artem

winder, 2016-05-14:
I moved those messages to the verbose console. It looks like something else might be going on with your installation though, because the MPos/WPos messages shouldn't have been picked up here.

Do things seem to be working other than all the extra messages in the console?

closed 2016-05-14 by winder

tema2121, 2016-05-14:
After starting sending a file first such message in the console stopping motion. Can I sending that  MPos/WPos messages back to verbose by typing a command or making change in log files?

winder, 2016-05-14:
The `show verbose output` checkbox above the console toggles whether these messages are displayed.

tema2121, 2016-05-15:
it it some thing with EL capital of Java

winder, 2016-05-16:
I've tested UGS with el capitan. I'm not sure whats causing this, that code hasn't been changed in several months.

tema2121, 2016-05-16:
I found that article https://oliverdowling.com.au/2015/10/09/oracles-jdk-8-on-mac-os-x-el-capitan/
and repeat all - but still that bug

commit 3213298f (2016-05-14): Move GRBL feedback messages to verbose console. Closes #415