winder/Universal-G-Code-Sender #416: Random switching from inches to millimeters

OldSalt1945, 2016-05-18:
New, old issue.  I contacted you on a different forum a few months back concerning imperial/metric problem.

I never use metric.  When jogging, occasionally and for no discernible reason,  UGS will switch to metric movements, even though the "inches" button is marked.  The only way I have been able to return to inches is to check the "milimeters" button, move an axis, move it back, check the "inches" button.

Additionally, when runnings the same job multiple times, it also changes to millimeters and machines a very tiny part.

I am running Version 1.0.9 / Nov 11, 2015.

I am a complete Noob at this.  Is there anything that I may be doing wrong?
The problem is sufficient that I have started using Easel for running my gcode, even though it is severely limited in it's capabilities.

You have produced a great product and have priced it VERY competitively.

John

winder, 2016-05-18:
John, I'm glad you appreciate the competitive pricing model :)

These issues should be resolved in the nightly build, could you give it a try and let me know if the behavior is still unexpected?

As for the part being changed to millimeter mode, that is probably because of an invalid g-code being sent which causes GRBL to ignore the units setting. UGS has better handling for that in the nightly as well, but it will require you to make sure the invalid gcodes are removed from your gcode file.

OldSalt1945, 2016-05-19:
Thanks.  I'll give it a shot and let you know.

Ordinarily, I avoid anything that is labeled "nightly build" or "here, give this a try".  I lean more to things labeled "Stable Version".  :)  :)  :)

OldSalt1945, 2016-05-26:
I just got a chance to run a job on the V 2.0 nightly / May 19, 2016.

Kudos, dude.  It popped on a "Unsupported Command" just like you said.
This is great!  This was on a T1M6 command, which I edited out of the gcode file.
It could just as easily have been a corrupted communications error which previously would have been discarded, but would now be stopped and flagged, enabling debugging to commence and a controlled restart to be employed.

With verbose turned on, does it also check to see if sent equals echoed?

John

OldSalt1945, 2016-05-26:
Let me rephrase that last sentence.

With verbose turned on, is there a comparison of "Command" and "Original Command" with a transmission errors flagged and work stoppage?

John

winder, 2016-05-26:
@OldSalt1945 There is the command table, you can see the ok/error responses there. That table has really poor performance though so I'm planning to remove it. Is that what you were looking for?

closed 2016-05-26 by winder

OldSalt1945, 2016-05-26:
I just got my answer, sort of.
UGS acts as I was hoping it would.  
Unfortunately, the Arduino board has a 16U2 USB to RS-232 communication chip that has bad firmware that I have been unable to reload.  It cause random characters to be dropped/changed.

UGS seems to catch that.

The following is a case in point.  This job is 238,329 gcode lines long.  I seldom can run it without something going wrong.  

I ran it twice.  The first time was successful.  The second time something went wrong, 
UGS popped an error and stopped, but the results are very confusing.

X257.374Y201.038Z-0.517	X257.374Y201.038Z-0.517	true	true	ok
X257.363Y201.047Z-0.516	X257.363Y201.047Z-0.516	true	true	ok
X257.355Y201.053Z-0.516	X257.355Y201.053Z-0.516	true	true	ok
X257.349Y201.058Z-0.516	X257.349Y201.058Z-0.516	true	true	error: Expected command letter
X257.335Y201.069Z-0.515	X257.335Y201.069Z-0.515	true	true	ok
X257.328Y201.075Z-0.516	X257.328Y201.075Z-0.516	true	true	error: Invalid gcode ID:25
X257.077Y201.287Z-0.519	X257.077Y201.287Z-0.519	true	true	error: Expected command letter
X256.996Y201.360Z-0.517	X256.996Y201.360Z-0.517	true	true	error: Invalid gcode ID:25
X256.981Y201.373Z-0.516	X256.981Y201.373Z-0.516	true	true	error: Invalid gcode ID:25

I found the commands in gcode using Notepad, but they are reversed with respect to the command log:

X256.961Y201.393Z-0.517
X256.967Y201.387Z-0.517
X256.981Y201.373Z-0.516
X256.996Y201.360Z-0.517
X257.077Y201.287Z-0.519
X257.328Y201.075Z-0.516
X257.335Y201.069Z-0.515
X257.349Y201.058Z-0.516
X257.355Y201.053Z-0.516
X257.363Y201.047Z-0.516
X257.374Y201.038Z-0.517
X257.380Y201.034Z-0.517
X257.492Y200.957Z-0.517
X257.567Y200.914Z-0.511
X257.615Y200.892Z-0.503
X257.640Y200.883Z-0.496
X257.650Y200.880Z-0.493
X257.661Y200.880Z-0.487
X257.676Y200.881Z-0.477
X257.702Y200.886Z-0.459

Which leads me to the response "Huh?"

Just to be thorough, I went to the beginning of the gcode file and found:

G21
G17
G0Z1.000
G0X0.000Y0.000S15000M3
G0X153.797Y2.952Z1.000
G1Z-1.687F720.0
G1X153.808Y2.952Z-1.687F2520.0
X155.681Y3.011Z-1.687
X155.687Y3.012Z-1.687
X155.703Y3.012Z-1.687
X157.571Y3.095Z-1.687

Which is as it should be.

And at the end:

X161.558Y13.624Z-0.686
X160.970Y13.231Z-0.337
X160.405Y12.856Z-0.000
X160.970Y13.231Z-0.337
X161.558Y13.624Z-0.686
X162.142Y13.333Z-0.353
X162.757Y13.030Z-0.000
X162.142Y13.333Z-0.353
X161.558Y13.624Z-0.686
G0Z1.000
G0X0.000Y0.000
G0Z1.000
G0X0.000Y0.000
M30

Which is also as it should be.

Color me confused.

John

OldSalt1945, 2016-05-26:
Just a minor little suggestion:  "Return to Zero" would be better if it would raise to the highest Z that is has encountered, move to X0, Y0, then go to Z0.  Less likely to do damage.