winder/Universal-G-Code-Sender #418: (suggestion) Modify work 0 control

jahnj0584, 2016-05-20:
Sometimes Im doing very precise work, and have to change the work 0 a few times to have the program line up perfectly with my workpieces. It really sucks having to have the machine jog from the operation, to home, then back to 0 then jog it by DeltaX/Y/Z and resetting home, then trying again. Is there any way of setting work 0 to be 272.2 instead of 278 in the console, or can you add this functionality?

winder, 2016-05-23:
Do you have a suggestion for what this sort of panel would look like?

I'm hoping to get my machine upgraded with limit switches so I can get used to these sorts of workflows, but for now it just isn't something I do so it hasn't been implemented.

You can do these things manually, this document is where I'll start when I get there:
http://www.shapeoko.com/wiki/index.php/G-Code#Using_the_Work_Coordinate_Systems

jahnj0584, 2016-05-23:
could be something as simple as double clicking the work 0 field, and being allowed to type in a new value, or having a modifier option where you can +/- Xmm from the old set work 0. 

The wiki basically says to move to it, then run the console command to reset. I want to do this without physically moving back to the 0, which is what Ive been doing. 

(if 0 is 1mm too far on X, i'll return to zero, jog 1mm for -x then reset the work zero. The issue is with the more precise things Im doing it's down to 0.1mm or less, so theres lots of "air cutting" to make sure its in the right spot.)

jahnj0584, 2016-09-28:
This was a silly request by me. The G92 command does this already! (but has its own bug, already submitted though!)

closed 2016-09-28 by jahnj0584