winder/Universal-G-Code-Sender #420: Make XYZ Work Position User Modifiable

OldSalt1945, 2016-05-26:
This is along the lines of a previous posting "(suggestion) Modify work 0 control #418".

If you were to make "Work Position" user modifiable, similar to what Mach 3 does, it would open up additional capabilities for the user.

This is typically done with probing fixtures.  Move Z down to touch the "Z plate", enter .250 to account for the thickness of the plate.  Move X to the edge of the fixture, enter .125 to account for the radius of the bit.  Etc.

Another use would be centering.  Move the cutter to the center of the work piece, then enter the XYZ coordinates for gcode that starts from the lower left corner.  Jog up some and click "Return to Zero".

I would think this would be a rather modest change.  You should have it done by lunchtime tomorrow. :)

John

OldSalt1945, 2016-05-26:
With this enhancement, you could double your price and get no complaints

jahnj0584, 2016-05-26:
you can already do that already, take the probe macro from chillipeppr, and just do a jog of 1/2 the bit diameter. This improvement would still help both of us though! I think winder cares more about HOW to let us change the work 0's.

winder, 2016-05-26:
I'm happy to add more features like this, I'm just not sure what the UX should be to make it easy to use. Some sort of coordinate management panel is high on the list of things I want to implement once I have limit switches hooked up on my machine.

You guys are both probably doing more advanced things with your machines than I am with mine, so I'd be curious to see what you guys thing a workspace management panel would look like.

jahnj0584, 2016-05-26:
would adding another bubble besides the jog command work or be possible? [As well as one for just the Z axis?] You can then change between a jog size, or a machine 0 size to change.

winder, 2016-05-26:
So you'd use the jog buttons to modify the work location? It's an interesting idea, but I'm not sure how intuitive it would be.

I don't think I'd change the jog panel for this. That panel does a very specific thing and I think that makes it easier to use especially for novice users. This sort of use case is a bit more advanced, so I think it deserves its own panel. It could also manage things like WCS, and maybe other commands documented here:
http://www.shapeoko.com/wiki/index.php/G-Code#Using_the_Work_Coordinate_Systems

jahnj0584, 2016-05-26:
to be honest, I dont use Work offsets. Everything has its own 0, and I like to go there manually and run the operation, move, 0, load the next file. Can you copy the current jog panel, but have it effect work 0 then? That would be a happy medium; It can just be another tab that the advanced users will pull out, and the newbies will hide away.

OldSalt1945, 2016-05-26:
"""""You guys are both probably doing more advanced things with your machines than I am with mine, so I'd be curious to see what you guys thing a workspace management panel would look like.""""""

Not with programming, we are not.

What I'm suggesting is:
Under "Machine Status",
Make "Work Position: X:  and  Y:  and  Z:  user modifiable.

Ordinarily USG would control those values, but, if the user wanted to, for whatever reason, he could change it to anything he wanted to, rather than just to zero.  You might even allow such a change during a pause of execution.  Don't know why someone would want to, but why not?

This should take, I would think, very little additional programming.  The machine would not move, the user would just have the additional ability to control the environment.

OldSalt1945, 2016-05-26:
> once I have limit switches hooked up on my machine.

A word of advice,  use the normally closed option on the switches.  Which means, that the switch is normally closed and conducting 5 volts.  When the switch is pressed, the switch is open and no longer conducts the 5 volts.  If you do it the other way around (normally open), it is very susceptible to electrical noise and you will get false tripping, which will ruin your run.  I did it the wrong way and ended up disabling the homing for a while I redid them.

John

jahnj0584, 2016-05-26:
John, i have mine as NO and had issues, all you have to do is disable hard limits, but homing still works. When I get the solder iron out next time I'll swap em to prevent any crashes or issues while cutting.

Winder, unless you already ordered them, I have 7 spare limit switches I'd love to send you.

OldSalt1945, 2016-05-26:
I've been trying to think of a "fer instance" of why this change might be useful.

So here goes:

Suppose you have a sign board that you have carved on your machine that took 3 hours to carve.
Now suppose that you need to add a logo in a very specific spot on that sign board.  There are several ways to accomplish this, but one of the easiest is to look at the logo and find a distinctive spot on the logo that can be aligned with the sign board.  Jog the router bit to the exact spot where you want that part of the logo to be carved.  Find the X,Y,Z of the logo spot within the logo drawing.  Enter those coordinates into UGS.  Send it.  Home might be inches away, but the logo will be carved with proper alignment to the spot chosen on the sign board.

John

rugbymauri, 2016-05-26:
Hi Will, perhaps the solution is a feature i was thinking about a time a go.
My idea was to extend the macro functionality to allow to enter a user input/value, call it "ValueMacro" or what ever. So to solve Johns problem for example setting the X to 123, the marco will be "G92 X123".
The "ValueMacro" could be "G92 X{New_X_Value}", UGS Parse the macro if it finds a "{....}" token, it ask for the value for "New X Value".

Mauri

jahnj0584, 2016-05-26:
John, it'd be easier in that case to just edit the work 0 of the CAD program to be in the spot where you can easily locate it. I'm offering the other idea, where you would air-cut the new logo, and measure how far off it is from where it should be and be able to change that in the software instead of physically.

OldSalt1945, 2016-05-26:
I am using V-Carve.  Origin is top or bottom of one of the 4 corners or center.  I have no interest in learning anything more complex.

What I am proposing is something that would add additional capability without additional complexity.

The term "Macro" should never be used in a context discussing "novice user".  Kudos to anyone who understands what it means, but no thanks.

I'm 71.  I don't have the patience to putz around with more new concepts.

I was a mainframe computer programmer for about 40 years.  I would think that the programming involved in what I propose would be:

- Modify "Machine status" "Work Position X: and Y: and Z: to be user input fields in addition to being output fields.

- If machine.is.at.rest (paused or stopped) and

- If screen.work.position not = work.position

- .......then work.position = screen.work.position

John

winder, 2016-05-26:
Thanks for all the input everyone.

@OldSalt1945 To your comment about new concepts - that's exactly the reason
I'm hesitant at all. If I were to add every idea then UGS then it would be
a much more complicated tool. So I'm cautious about how they are added.

OldSalt1945, 2016-05-26:
@winder To your comment about new concepts - that's exactly the reason
I'm hesitant at all. If I were to add every idea then UGS then it would be
a much more complicated tool. So I'm cautious about how they are added.

It would be easy to include topsy turvy additions until it looked like chillipeppr.

I suspect that clillipeppr is more capable than UGS, but so complex that I am unwilling to spend the mental capital to learn it when, as a person with only the experience I have from building a wooden CNC machine and learning about it from youtube videos, I don't know what to expect.

Easel is more of my speed, but so VERY lacking in capability, that I am willing to deal with a "little" more complexity.  UGS fits my needs exactly.......................but could you change just this one little thing??????????????  :):):):):):):)

winder, 2016-06-18:
@OldSalt1945 I implemented the suggestion @rugbymauri suggested. I know you were hoping to avoid macros, but it now allows you to make a button that does what you are asking for:

`G10 P1 L20 X{machine_x} Y{machine_y} Z{machine_z}`

I also added substitutions for:
`{work_x}, {work_y} and {work_z}.`

Plus a modal substitution that prompts you for the value to substitute:
`{prompt|<name of value>}`

closed 2016-06-18 by winder

OldSalt1945, 2016-06-18:
That's OK.  I still like you.  ( :-) )==

John

jahnj0584, 2016-06-18:
so we just click the macro, then type in the new X0 value (in relation to machine 0 I assume?)

Thanks fam

winder, 2016-06-19:
@jahnj0584 yes, here's an example:
`G0 X{prompt|X location} Y{prompt|Y location} Z{prompt|Z location}`

When you run the macro:
[image]

The command that gets sent:
[image]