winder/Universal-G-Code-Sender #421: How to set GBRL firmware settings via the new Platform UGS ?

splitn2, 2016-05-29:
Hi,

I set up a small cnc on the weekend with GRBL 0.9j , and its working well.however I could only find firmware setting feature working in the older UGS 1.09 but when I tried in latest downloaded Platform UGS there were no settable firmware features to set the following.....

$0=10 (step pulse, usec)
$1=25 (step idle delay, msec)
$2=0 (step port invert mask:00000000)
$3=6 (dir port invert mask:00000110)
$4=0 (step enable invert, bool)
$5=0 (limit pins invert, bool)
$6=0 (probe pin invert, bool)
$10=3 (status report mask:00000011)
$11=0.020 (junction deviation, mm)
$12=0.002 (arc tolerance, mm)
$13=0 (report inches, bool)
$20=0 (soft limits, bool)
$21=0 (hard limits, bool)
$22=0 (homing cycle, bool)
$23=1 (homing dir invert mask:00000001)
$24=50.000 (homing feed, mm/min)
$25=635.000 (homing seek, mm/min)
$26=250 (homing debounce, msec)
$27=1.000 (homing pull-off, mm)
$100=314.961 (x, step/mm)
$101=314.961 (y, step/mm)
$102=314.961 (z, step/mm)
$110=635.000 (x max rate, mm/min)
$111=635.000 (y max rate, mm/min)
$112=635.000 (z max rate, mm/min)
$120=50.000 (x accel, mm/sec^2)
$121=50.000 (y accel, mm/sec^2)
$122=50.000 (z accel, mm/sec^2)
$130=225.000 (x max travel, mm)
$131=125.000 (y max travel, mm)
$132=170.000 (z max travel, mm)

any ideas how to do this in the latest Platform or maybe I did something really dumb wrong, but I am stumped......

cheers

winder, 2016-05-30:
@splitn2 It's not you, I haven't added the GRBL settings GUI to the platform yet. It's on the short list of features I need to finish before releasing 2.0. For now you can still do it the old fashioned way by sending "$$" to see the options then setting them directly with $##=value.

closed 2016-05-30 by winder

splitn2, 2016-05-30:
@winder thanks I hadn't even got to trying it the old fashioned way!!! By the time I barely got it going last night the whole system was whisked away back to his home town 500km away, we are going to have another setup session via screen sharing so I can tweak settings remotely. Thanks again Will great work! Hey final question for now, when GRBL 1.x is released is UGS going to run fine with it out of the box? cheers