winder/Universal-G-Code-Sender #422: Missing settings on $$ command

TheAHan, 2016-06-02:
I have an issue where my list of available settings (on $$) is shorter than it should be, most notably for me it's misssing the x,y and z settings:
[image]
Is this an issue with GRBL rather than UGCS, and does anybody know what the fix could be?

winder, 2016-06-02:
If you enable `Show verbose output` are the other commands displayed?

TheAHan, 2016-06-02:
I will have to check later tonight when I get home. I've been reading on the GRBL github here: https://github.com/grbl/grbl/wiki/Frequently-Asked-Questions that the settings list is stored in the Arduino EEPROM, and that sometimes settings and parameters are funky after flashing, so I'm going to try that tonight too.

Is this most likely a problem with GRBL though?

TheAHan, 2016-06-04:
So yes, I sent a $$ command with verbose output enabled and the settings are there:
[image]
It turns out I can also edit them and they will save normally.
Why are they verbose?

bobdewar, 2016-06-08:
I am also only seeing settings 0 to 26 even if I have select "show verbose output".  It works fin in Version 1.0.9.  I get the same result if I enter "$$" in the command window or view the options using Settings/Firmware Settings/GRBL.

Here is what I see;

>>> $$
$0=10 (step pulse, usec)
$1=255 (step idle delay, msec)
$2=0 (step port invert mask:00000000)
$3=0 (dir port invert mask:00000000)
$4=0 (step enable invert, bool)
$5=0 (limit pins invert, bool)
$6=0 (probe pin invert, bool)
$10=3 (status report mask:00000011)
$13=0 (report inches, bool)
$20=0 (soft limits, bool)
$21=0 (hard limits, bool)
$22=0 (homing cycle, bool)
$23=0 (homing dir invert mask:00000000)
$26=250 (homing debounce, msec)
ok

barnold96, 2016-06-08:
All I get with "show verbose" on is the following.  There's a constant output of lines as below, then only 0-26 when I enter $$.

[verbose]<Idle,MPos:0.0000,0.0000,0.0000,WPos:0.0000,0.0000,0.9999>
[verbose]<Idle,MPos:0.0000,0.0000,0.0000,WPos:0.0000,0.0000,0.9999>
[verbose]<Idle,MPos:0.0000,0.0000,0.0000,WPos:0.0000,0.0000,0.9999>
[verbose]<Idle,MPos:0.0000,0.0000,0.0000,WPos:0.0000,0.0000,0.9999>
[verbose]<Idle,MPos:0.0000,0.0000,0.0000,WPos:0.0000,0.0000,0.9999>
>>> $$
$0=10 (step pulse, usec)
$1=255 (step idle delay, msec)
$2=0 (step port invert mask:00000000)
$3=3 (dir port invert mask:00000011)
$4=0 (step enable invert, bool)
$5=0 (limit pins invert, bool)
$6=0 (probe pin invert, bool)
$10=3 (status report mask:00000011)
$13=1 (report inches, bool)
$20=0 (soft limits, bool)
$21=0 (hard limits, bool)
$22=0 (homing cycle, bool)
$23=0 (homing dir invert mask:00000000)
$26=250 (homing debounce, msec)
ok

CritterCJ, 2016-07-25:
I have the same issue. List of setting always ends $26. Last version of GRBL Gcode sender where it works is about 2.0 (nightly) / Mar 28, 2016. Newest versions always have this issue.

Settings are missing too. See attached image.

[image]

commit f6b5c974 (2016-07-31): Fix grbl feedback/settings regex. Closes #422
closed 2016-07-31 by winder