winder/Universal-G-Code-Sender #426: release of V2?

jahnj0584, 2016-06-21:
So, whens the big reveal? We've been having lots of pushes which is great, but whats all left before you'll release it as 2.0? If you could also add a "last updated' line after the downloader, that'd be great.

winder, 2016-06-22:
It's getting there, I try to update the changelog when something big has been added. Most of the really challenging parts are finished but there are still a bunch of little things left on my todo list:

- [ ] port remaining gui windows from netbeans GUI builder to MigLayout.
- [ ] grbl settings gui
- [ ] Fix macro settings gui
- [ ] Fix visualizer / gcode editor layouts, and line highlights when windows are moved.
- [ ] Configurable gcode preprocessors.
- [ ] Gcode preprocessors to ignore common unsupported gcode commands (`T*`, `G64`, others?).
- [ ] GUI for starting and connecting to the Pendant web interface.
- [ ] Auto-update.
- [ ] Replace console text area with the built in netbeans console component.
- [ ] Update `Workflow Plugin` documentation to include Maven and localization.
- [ ] Fix splash screen and program icons to work with Maven.
- [ ] Integrate GSharp preprocessor into editor.
- [ ] Hope for some volunteers to localize the platform: https://poeditor.com/join/project/2J2hB5I41Z

closed 2016-06-22 by winder

jahnj0584, 2016-06-22:
Dont forget the M commands! T and M are the only ones that have not agreed with UGS so far for me.

I have a pretty nice color UI set for my visualizer if you want me to share it. It might be good to ship with a white and grey layout out of the box.