winder/Universal-G-Code-Sender #427: TinyG can't set work Zero

Splocal, 2016-06-22:
I've had a problem setting work Zero while using my TinyG cnc controller. I can jog the machine but even manually inputting G0X0 for instance does not move my axis to zero. Does anyone else have this issue with TinyG?

winder, 2016-07-19:
Do other commands work? Can you send a file? Anything in the console when sending a command?

closed 2016-07-31 by winder

Splocal, 2016-07-31:
I tried multiple commands input manually some would work and others nothing.

winder, 2016-07-31:
You could try using JSON wrapped commands like `{"gc":"g0 x0"}`. Described on the TinyG wiki: https://github.com/synthetos/TinyG/wiki/JSON-Details

Splocal, 2016-08-01:
Thanks, I'll give that a try. Is this a known Issue with UGS and TinyG? I was hoping to use UGS for my larger Gcode files as Chilipeppr locks up when my file is too large.

Splocal, 2016-08-01:
I am thinking I can zero my machine with chilipeppr and then Load a file in UGS and let it run. I won't have all the functionality of chilipeppr but if it can successfully complete a Gcode file than I would be satisfied with it.