winder/Universal-G-Code-Sender #428: (suggestion) Current speed display

jahnj0584, 2016-06-23:
velocity, speed, whatever you want to call it. It'd be nice to see what the machine is ACTUALLY moving at. I feel like even if i do something at 4k or 3.5k its moving at the same rate (or maybe i have a limit somewhere?) so it'd be cool to see what the real rate is! Just as easy as DeltaX/1 second * 60 = Xmm/min.

I guess I could run the thing twice, and compare the times but its easier for you to do it ;) 
Cheers

rjclocks, 2016-07-06:
It may be far simpler (without any of that pesky mathematics) to show the last feed rate setting (Fnnnn).
Although this won't work for any G0 commands.

jahnj0584, 2016-07-06:
That's a good idea, but when speed override is finally functioning it won't
help much that way.

jahnj0584, 2016-09-23:
Any word, will?

winder, 2016-09-23:
I'm revamping the controller status code to support a some new features coming soon in grbl 1.0, I'll keep this in mind while doing that.

jahnj0584, 2016-09-23:
Great. Did you get a chance to look at my Java issue? I can't use platform
anymore

jahnj0584, 2016-09-28:
Any idea on when this will be implemented? I'd prefer this to backlash compensation (which really isnt necessary for CNC's)

faultylee, 2016-09-28:
I thought backlash compensation should have been in the firmware instead of on the UGS, it's more closely related to the machine

jahnj0584, 2016-09-28:
It will be, but Will would have to spend time to code in the program to choose the backlash amount and all that jazz.

winder, 2016-09-28:
The next version of GRBL which is being released soon has the current speed (along with some other metadata) sent in the status messages. So this is only going to be supported with that version of GRBL - but it will be supported soon!

I haven't thought about backlash compensation in a while. Now that the Gcode Parser has modular processors there could be a simple module that adds in backlash compensation. It might not even be very difficult to add.

supagu, 2016-11-11:
it would be good to have a percentage/time till completion, elapsed also.

jahnj0584, 2017-09-04:
is this currently working in the platform? My machine wont be running until next week so I cant really check.

breiler, 2018-08-08:
This has been possible for a while, so I'm closing this. Now it also converts the feed speed to the preferred units: #1105

closed 2018-08-08 by breiler