winder/Universal-G-Code-Sender #430: (suggestion) Numbers in visualizer are always normal-to the viewer

jahnj0584, 2016-06-30:
sucks having to always rotate back to the flat views to see the dimensions. It'd be nice if the numbers DIDNT rotate with the viewer, and maybe even had an option to show them outside of the constraint lines? 

Add-On: allow clicking on a face of the XYZ cube to instant snap to that view. 
[image]

winder, 2016-09-07:
After implementing this, I think that it looks pretty bad/confusing when enabled for the X/Y axes. Maybe I'll enable it just for the Z axis.

[image]

[image]

closed 2016-09-07 by winder

jahnj0584, 2016-09-07:
Works for me! The Z one was the one that got the messiest anyways.

commit 06016171 (2016-09-07): Rotate SizeDisplay text for the Z axis so that it always faces the camera. Closes #430