winder/Universal-G-Code-Sender #433: send file run away

weirdguy79, 2016-07-03:
getting frequent random run aways when sending a file, cancel soft reset then send again and off it goes carving with no problem

winder, 2016-07-04:
When you start the machine do you check that the coordinates are at 0,0,0?
If you are using the reset zero button you need to do that before every
send.

closed 2016-07-19 by winder

weirdguy79, 2016-07-19:
coordinates are at 0,0,0 reset used before every send, has done it in every version I've used so far

winder, 2016-07-19:
Could you describe the steps you typically take when seeing this issue? In as much detail as you can, there is probably something specific about your workflow that isn't being accounted for.

Also what version of UGS are you using?

weirdguy79, 2016-07-21:
I load a file, move to desired position start position, hit reset zero, hit send, sometimes the cnc just starts to make random moves that have no relation to the file at all, hit cancel, move back to home, hit send again and off it goes carving as intended. am using latest version but have had the issue ever since i have used UGS (about a year or so), seems to be happening more frequently these days