winder/Universal-G-Code-Sender #434: Not loading all GRBL Settings

jahnj0584, 2016-07-04:
[image]

winder, 2016-07-04:
If you send another $$ does it show all the commands?

winder, 2016-07-06:
I finally got a chance to try this out, for some reason these messages are being marked as verbose. So they are filtered out. I'll need to take a closer look at whats going on here.

```
$0=10 (step pulse, usec)
$1=25 (step idle delay, msec)
$2=0 (step port invert mask:00000000)
$3=0 (dir port invert mask:00000000)
$4=0 (step enable invert, bool)
$5=0 (limit pins invert, bool)
$6=0 (probe pin invert, bool)
$10=3 (status report mask:00000011)
[verbose]GrblFeedbackMessage{message='$11=0.020 (junction deviation, mm)', distanceMode='null', units='null'}
[verbose]GrblFeedbackMessage{message='$12=0.002 (arc tolerance, mm)', distanceMode='null', units='null'}
$13=0 (report inches, bool)
$20=0 (soft limits, bool)
$21=0 (hard limits, bool)
$22=0 (homing cycle, bool)
$23=0 (homing dir invert mask:00000000)
[verbose]GrblFeedbackMessage{message='$24=25.000 (homing feed, mm/min)', distanceMode='null', units='null'}
[verbose]GrblFeedbackMessage{message='$25=500.000 (homing seek, mm/min)', distanceMode='null', units='null'}
$26=250 (homing debounce, msec)
[verbose]GrblFeedbackMessage{message='$27=1.000 (homing pull-off, mm)', distanceMode='null', units='null'}
[verbose]GrblFeedbackMessage{message='$100=43.750 (x, step/mm)', distanceMode='null', units='null'}
[verbose]GrblFeedbackMessage{message='$101=43.750 (y, step/mm)', distanceMode='null', units='null'}
[verbose]GrblFeedbackMessage{message='$102=320.000 (z, step/mm)', distanceMode='null', units='null'}
[verbose]GrblFeedbackMessage{message='$110=5000.000 (x max rate, mm/min)', distanceMode='null', units='null'}
[verbose]GrblFeedbackMessage{message='$111=5000.000 (y max rate, mm/min)', distanceMode='null', units='null'}
[verbose]GrblFeedbackMessage{message='$112=500.000 (z max rate, mm/min)', distanceMode='null', units='null'}
[verbose]GrblFeedbackMessage{message='$120=25.000 (x accel, mm/sec^2)', distanceMode='null', units='null'}
[verbose]GrblFeedbackMessage{message='$121=25.000 (y accel, mm/sec^2)', distanceMode='null', units='null'}
[verbose]GrblFeedbackMessage{message='$122=10.000 (z accel, mm/sec^2)', distanceMode='null', units='null'}
[verbose]GrblFeedbackMessage{message='$130=200.000 (x max travel, mm)', distanceMode='null', units='null'}
[verbose]GrblFeedbackMessage{message='$131=200.000 (y max travel, mm)', distanceMode='null', units='null'}
[verbose]GrblFeedbackMessage{message='$132=200.000 (z max travel, mm)', distanceMode='null', units='null'}
ok
[verbose]GrblFeedbackMessage{message='[G0 G54 G17 G21 G90 G94 M0 M5 M9 T0 F0. S0.]', distanceMode='G90', units='G21'}
```

winder, 2016-07-06:
#422 is the same issue.

closed 2016-07-06 by winder

CritterCJ, 2016-07-22:
I have the same issue. List of setting always ends $26. Last version of GRBL Gcode sender where it works is about 2.0 (nightly) / Mar 28, 2016. Newest  versions always have this issue.