winder/Universal-G-Code-Sender #440: No "Firmwares"

barnold96, 2016-07-19:
I downloaded the latest build, 595, this morning.  When I ran it, I got this window.

[image]

winder, 2016-07-19:
What OS are you using? Are there any errors on the console?

I made some changes to how the different firmwares are configured, looks like the code to initialize those settings isn't working on your machine.

barnold96, 2016-07-19:
Win 10.  Also fails to connect after clicking past the box.

--------------------------------------------

 

 
 
 it on GitHub, or mute
 the thread.

winder, 2016-07-20:
I found the issue last night but wasn't able to finish fixing it, hopefully tonight.

commit 3334a872 (2016-07-21): Fix firmware config file jar extraction. Closes #440
closed 2016-07-21 by winder