winder/Universal-G-Code-Sender #441: no port selection

jahnj0584, 2016-07-20:
[image]
I had to add the connect button through the options

winder, 2016-07-20:
I'm pretty sure this is a side effect of #440

jahnj0584, 2016-07-20:
ah good point. Didnt think an issue in one would effect the other platform. Whens the fix, boss?

winder, 2016-07-20:
Yeah, probably more than 90% of the code is shared between the 2 GUIs. If I run out of time tonight I'll definitely get it on the train tomorrow morning.

jahnj0584, 2016-08-04:
[image]
I think the issue is still there after the aug 1st update will

winder, 2016-08-04:
If you right click in the toolbar area is there a `Connection` option? I think you might have just disabled it on accident.

jahnj0584, 2016-08-04:
Nope. this is how it is with a fresh download, Macros' arent there from my old version either

[image]

closed 2016-08-04 by winder

winder, 2016-08-04:
@jahnj0584 I found the problem, it should be fixed now. This was a really strange issue that wasn't reproducible from my development environment so I never noticed it. Thanks for reporting the problem.

commit 2a433a78 (2016-08-04): Fix resource loading from nested jar. Closes #441